betaflight/blackbox-log-viewer #165: Show amount of vertical scaling, expo and smoothing on-screen

ctzsnooze, 2018-05-05:
It would be helpful if the amount of the zoom, expo and smoothing settings was present on the screen somewhere.  

The reason is that people are always making screen images but I don't know what these settings are, so I can't interpret the image.  

It is difficult because these settings can vary trace by trace, and the amount of info is not trivial.

Eg smooth 50% / expo 40%/ zoom 200% might be represented by the string ```s50e40z200```

I don't know where that string would best appear, because there is not enough room on the legend area, which is where it logically would go.

Also the complication is that the z key does a time zoom...

Perhaps this info could be shown optionally at hard left, behind the waveform area, in the colour of the trace?  It could be toggled on and off like the 'craft' icon, but be on by default, so new users would know what the scaling, vertical  actually was at.

wolkesson, 2018-05-05:
Great idea! It's important info when trying to help out using screenshots. If we show the smooth, expo and zoom per trace, the magic mouse wheel scroll interactions and overrides would be a lot easier to understand. I would suggest squeezing it in the legend anyway. 

We could move the log number selection dropdown to the empty space below the legend. The header "Legend" could go away. That should give us negative space, which is positive I this case. :) Then I think we can fit the numbers after the name on the legend in a faded color to not draw too much attention. Making it fit is a challenge I'm willing to accept... B-) I'll be making a paint mock-up this weekend!

wolkesson, 2018-05-13:
Here's a paint mock-up! Either; 
A) we do as on the first graph and put it between the name and value, but that means that we might have to increase the width of the legend a bit to fit the longest names. And maybe find another solution for the "analyze button". 
B) we do as on the last field and put it on a wider line below the field name and value. 
C) we put it on as cryptic numbers on the actual plot area. 

[image]

A will take up more space horizontally and B might limit the number of simulations fields. But, in my opinion, there is no use case where you need to plot 20+ fields on a standard 1920x1080 screen. So horizontal space is more valuable and hence B is better than A. One downside of both A&B would be that neither will show when exporting a video. C is on the other hand very cryptic and the normal user won't know what it is. On the legend we could also add mouse interactions for less cryptic scaling.

@McGiverGim , @mikeller : What's your opinion on this?

McGiverGim, 2018-05-13:
The idea es very good, but I don't know where to place it to fit perfectly.

About this, I've pushed a PR to "give more user friendly names" to the legend, and thus they are a bit longer, so less space to this info. 

About A the good notice is that the legend adjusts the size automatically to the contents, so it will fit.

I like your B idea to put them in the line, but this width of line I think is less elegant... Sincerely I'm not too sure.

ctzsnooze, 2018-05-14:
To be honest I am thinking that the legend might get too wide with the numbers next to the name.

So I think on the far left looks good and gets the job done!  

Thank you!

wolkesson, 2018-05-14:
@mikeller: With A, I think you mean C (in the actual plot not the legend)? :)

mikeller, 2018-05-14:
@wolkesson: Yes, that's correct. ;-)

McGiverGim, 2018-05-31:
@wolkesson some advance with this? I think is the latest "great" feature left for the release of Betaflight 3.4. It seems that B will be enough by the moment.

wolkesson, 2018-06-02:
I’ve come about halfway with option B, but won’t be able to continue for some days... Will skip scrollwheel handling for now, and add it in later because it would be neat to scroll on the number to change that property.

McGiverGim, 2018-06-03:
As always, don't worry, when you have time. It's not a critical feature. Thanks for the update!

wolkesson, 2018-06-19:
I'm still working on this and I'm getting close now. I still have hover-highlight and mousewheel interaction to do. **Comments please!** Is it to large, unreadable, nice or just plain ugly? 
[image]

The idea is to have this on by default so that novice users (unknowingly) will provide the data needed for analysis by more experienced users. But maybe there should be an option to turn it off?

One could also argue for completely different solutions, but this is what we decided on above, so we do this first and then anyone is free to provide another solution.

McGiverGim, 2018-06-19:
I don't see a need to disable this. Let it on for all users.

What I see ugly is the text, I think is better `Z100 E25 S30` with some space between values than `z100e25s30` that seems a product code...

I see you have aligned text of the legend left, I think that is right. Maybe the zoom, expo and smooth are better in the left too?

For the rest, I see is good for a first version. If we find a better way after that we can change it then.

mikeller, 2018-06-19:
@wolkesson: This looks good to me too. And I agree with @McGiverGim, unless somebody comes up with a specific use case for disabling it, leaving it on permanently is fine. And I think adding spaces between the values is a good idea too, it will make it easier to read the small font.

ctzsnooze, 2018-06-20:
Agree, having it on always is great; the most visually nice way of doing it I leave to you guys :-)

PR #211 (2018-06-20): Extend the graph legend with zoom, expo and smooth info. 
Fixes #165.
closed 2018-06-20 by mikeller