betaflight/blackbox-log-viewer #167: Expo, Smooth and Grid buttons hide when enough space

McGiverGim, 2018-05-05:
If I made the window small enough, the zoom is converted into a menu and some "new" options appear:

[image]

These options are not available when the window has space enough. This can be fixed adding the buttons in the expanded form, in the same way than there are in the view menu.

I've done it, but my skills in design are very bad, so I ask help for this ;)

@wolkesson you did the reorder or the menu/design, and seems you have experience with this. Have you time to help me with this?

My idea is get rid or this other PR: https://github.com/betaflight/blackbox-log-viewer/pull/161 to do this, I need:
- Get a "decent" design of the new buttons/actual menu.
- Store this and other settings in the chrome storage, in this way they are visible and stored betweens sessions.

What do you think? New images for these "buttons" are the first thing we need.

wolkesson, 2018-05-05:
That would be a great change! I noticed this when I rebuilt the toolbar but decided to leave it just to keep the change small. But it definitely should expand into buttons when there is space enough. 

Question is if it should be on the zoom menu at all, since the zoom is time-wise not in amplitude? I'll create some icons (smooth, expo, scale, grid) and put them on this thread for discussion. I can then add dummy buttons to the toolbar after that. Then I'll hand over to you to implement the actual functionality. :)

McGiverGim, 2018-05-05:
Yes, any change in the UX that you can do will help a lot .The functionality is there, the actual menu items work. I can add later the "store" in the chrome storage without problem.

Thanks again for your help!!

wolkesson, 2018-05-09:
We suddenly go very nice flying weather! Ergo the time in front of the computer has been very limited... I've started it but will require more time to get the responsiveness right. 

I think we should also add a button for overriding Vertical Zoom as #166 suggests? I'd like to call this "scale" to not confuse it with a horizontal/time "zoom". I'll prepare this PR with commented out stuff, but the implementation will be done separately. Adding the button, later on, will mean I'll have to redo the whole layout breakpoint design all over otherwise. That's why I'll add a hidden button already now. 

Icons will be solid black when active (i.e. smoothing is applied) and greyed out when inactive (override is in effect). I will also change the wording in the drop-down to not be a double negative. Here are some ideas for icons:
Expo [image]
Scaling [image]
Smoothing [image]

McGiverGim, 2018-05-09:
Hi! Great images! I think they will do the job.

Don't worry about the time. When you have time, it will be enough.

I think is better to ignore https://github.com/betaflight/blackbox-log-viewer/issues/166 by the moment. The problems, one by one. There is no actually a way to do this, and I'm not sure if it will be useful enough: the vertical zoom is defined by two things, the zoom desired and a conversion hardcoded depending on the value that is being shown. This is done to try to put different units in the same graph. So maybe the changes needed are bigger than expected or the results are not good enough.

So I think is better to solve first the scaling/smoothing problem, make the PR and merge it, and, when done, take a look to this vertical zoom problem.

What do you think?

wolkesson, 2018-05-11:
Hi again! 
You're right, always solve one problem at a time!

I've come up with two alternative placements for the buttons; one is on the toolbar, the other is in the legend. Of course, we shouldn't have both. Since the setting only affects the graph I figured it might as well be in the legend. Also this way it doesn't have to collapse into the other dropdown when the screen width gets too small. 

[image]

I also think that we should remove the indications on the status bar, now that it's visible elsewhere. 

I prefer the buttons on the legend. What's your opinion? When decided, I'll clean it up and do a PR.

McGiverGim, 2018-05-11:
Hi! I like more in the toolbar, I don't like have to search in the screen where is the button. But is not my decision only. @mikeller what do you think?

The result is great in both places, good work!

mikeller, 2018-05-11:
I think both placements are good. If I had to decide I would go for putting them into the legend, because that solves the problem of the toolbar being overloaded, and makes sure that the information about the grid mode is included when users take screenshots of graphs, which is a nice side effect.

PR #175 (2018-05-23): Ui override in legend
Fixes #167 

Images are embedded SVG. 
Removed toolbar dropdown because buttons in the legend are not affected by resizing. 

Would you like me to also remove the status bar texts?
closed 2018-05-23 by mikeller