betaflight/blackbox-log-viewer #169: Blackbox/Video desync

McGiverGim, 2018-05-07:
I don't know if maybe is a problem with my video (Runcam Split 2), but I have observed that the video is slightly faster than the blackbox log.

If I sync the video with the log, and I advance the log, the video is faster. Maybe others had notice the same?

I don't know where the problem can be. First, is better to see if I'm the only one (is a problem of mine) or is a bug.

McGiverGim, 2018-05-13:
I think that this problem is a bug in the Runcam Split 2.

I've recorded a video with the Runcam Split 2. At the start of the video I've shown a chronometer, and the same chronometer at the end. The result is like this:
- The chronometer has advanced 7 minutes 14 seconds
- The runcam split has recorded, while this time, a video of 7 minutes 25 seconds

This 11 seconds of difference are not noticeable while you see the video, but when you sync it with the Betaflight Blackbox Log to analyze a flight, the log and the video desyncs in the time.

I think that the problem is with some internal clock of the Split, but I don't know if this can be fixed or not.

I will close this issue, but I think is a good idea to let this open 3 or 4 days to see if someone can confirm this.

McGiverGim, 2018-05-13:
@azolyoung can you confirm this problem with Runcam Split 2?

azolyoung, 2018-05-14:
OK, got it, I'll test it with split in tomorrow:)

McGiverGim, 2018-05-14:
Thanks! Waiting for your conclusions!

flint723, 2018-05-17:
I've seen the same with Fatshark DVR loaded into BBE.  Slight video desync by the end of the log.  I don't have exact numbers but I've noticed it a few times now.

McGiverGim, 2018-05-17:
If you can, make the same test than me: record a clock at the beginning and at the end and see if the time in any video player is correct with the clock difference. In my case it seems a problem with the Split, maybe in yours is with the Fatshark DVR ;)

19Gerhard85, 2018-05-18:
@flint723 
The Fatshark DVR is dropping frames if the reception is too bad (static).

flint723, 2018-05-18:
OK thanks, makes sense.

McGiverGim, 2018-05-23:
@azolyoung did you test the problem? Can you confirm it?

Lord-Quake, 2018-06-06:
Tested with build 75:
Loading a BT Log and a video from a Mobius Mini the log will gradually lag behind.
Example screen shot of a full roll:
[image]

McGiverGim, 2018-06-07:
@Lord-Quake please record a watch to see if this is problem of the cam. The same length as your sample, about 7 minutes.

Lord-Quake, 2018-06-07:
According to the test my 7 min video is off by 00;00;00;04 (hr;min;sec;msec). Checked using PowerDirector 14.

Lord-Quake, 2018-06-08:
@McGiverGim 
I've uploaded a 5min video for you to use as a test object:
http://www.leo.nutz.de/leo/tmp/5min_Video_Test.MP4
Let me know when you have downloaded it so I can remove it from the server.

Lord-Quake, 2018-06-09:
@McGiverGim  Fixed. Sorry about that!

McGiverGim, 2018-06-09:
Yes, the time in your video seems right. Can you upload some real video with some real log? But if it desync I suspect maybe something is wrong in the firmware.

Lord-Quake, 2018-06-09:
How do you want the video to be made?
Log overlapping the video using the BF viewer and saved as webm?

McGiverGim, 2018-06-09:
No, no. Two independent files, in this way I can play with it. One file with the video and other with the log.

Lord-Quake, 2018-06-09:
OK will do. Will do more testing this afternoon and then upload the files.

Lord-Quake, 2018-06-11:
I have a zip file. It's 1 gig in size. It's best you send me your email to leo@nutz.de 
I will then send you a link to download. I don't want the video to be made avail to the public.

McGiverGim, 2018-06-11:
Thanks! Send it to mcgivergim@gmail.com

McGiverGim, 2018-06-11:
@Lord-Quake I've been studying your video/log, there is a little of desync, about 600ms in 5 minutes 30 seconds of video. Is not the same case than mine (mine was about 20 times yours) but I will study it.

I don't know if the culprit is the video parser of the blackbox or the firmware clock writing the time into the log that is not perfect. I will try to get to some conclusion. Thanks for your video!!

Lord-Quake, 2018-06-11:
Yes, I'm seeing about 700ms @ 5:27.446. While that doesn't seem much it is very much noticeable when looking at the video playback. Good luck!

closed 2018-07-18 by stale[bot]