betaflight/blackbox-log-viewer #185: Gyro and Debug Misaligned

spatzengr, 2018-05-24:
In the master, as of 5/23 it appears the scaling of the gyro vs. debug modes is off.  See below images from 5/11 master vs. 5/23 master.

[image]

[image]

McGiverGim, 2018-05-24:
Can you attach our blackbox file? Thanks!

spatzengr, 2018-05-24:
Yes, I should have done that.  Apologies.

[2018.05.18-BF 3.4.0-Build 793_PT1=90+DyN+PT1=300_Dt=PT1-100,200-FIRST LOS.txt](https://github.com/betaflight/blackbox-log-viewer/files/2035448/2018.05.18-BF.3.4.0-Build.793_PT1.90.DyN.PT1.300_Dt.PT1-100.200-FIRST.LOS.txt)

McGiverGim, 2018-05-24:
Tested and works with the https://github.com/betaflight/blackbox-log-viewer/pull/186

closed 2018-05-24 by mikeller