betaflight/blackbox-log-viewer #192: Cannot open video .AVI files produced by Eachine VTx + DVR

hermanblarsen, 2018-05-28:
Link to Eachine VTx+DVR see [https://www.banggood.com/Eachine-DTX03-DVR-5_8G-72CH-025mW50mW200mW-Switchable-VTX-w-Audio-NTSCPAL-p-1141123.html?gmcCountry=GB&currency=GBP&createTmp=1&utm_source=googleshopping&utm_medium=cpc_ods&utm_content=heath&utm_campaign=pla-fpv-gb-pc&gclid=EAIaIQobChMIq93m5p2o2wIVibTtCh1lGgI8EAQYASABEgJzHfD_BwE&cur_warehouse=CN](url)

Produced file by the VTx for testing, ie opening successfully: [PICT0001.zip](https://github.com/betaflight/blackbox-log-viewer/files/2044745/PICT0001.zip)

Current Warning Message produced bt The betaflight Blackbox Explorer 3.0.0 on WIndows 10: 
[image]

I'm not sure if AVI is supported as there is no list of supported file formats - a list would be good! Converting AVI to .mp4 using Handbrake as described in Cleanflights ReadMe allowed me to load the mp4 file.

mikeller, 2018-05-28:
@hermanblarsen: The problem with 'supporting AVI' is that AVI is what is called a 'container format'. This means that it just describes how data blocks are arranged in the media stream, but the format that is used to encode the data is not specified by the AVI format - it can be pretty much anything. So without knowing what video format a particular device that records AVI files uses, it is impossible to tell if this format will be supported or not.

McGiverGim, 2018-05-28:
AVI is only a container and is supported by the Blackbox (it supports AVI, MOV, MP4, MPEG). The container includes inside the video in different codec formats. If something is wrong while loading the video then the error is thrown.

McGiverGim, 2018-05-28:
mikeller was fastest than me :)

What is a little strange is the error message. The latest part (Try Google Chrome instead) has no sense. Maybe is better to remove it and change it by a more generic codec video error.

mikeller, 2018-05-28:
Yeah, that message is weird - and we are generating it. Should definitely change this.

McGiverGim, 2018-05-28:
@hermanblarsen I've tested your AVI sample attached, and the error is a CODE 4: MEDIA_ERR_SRC_NOT_SUPPORTED.

So it seems chrome does not support your coded. At least by the moment we can't fix this.

closed 2018-05-28 by mikeller