betaflight/blackbox-log-viewer #195: Stick Display - Top is cut off (tested with build 72)

Lord-Quake, 2018-06-03:
It doesn't matter where it is positioned and what size it has.
[image]

McGiverGim, 2018-06-03:
@wolkesson maybe something here https://github.com/betaflight/blackbox-log-viewer/pull/188 can have this secondary effect?

Lord-Quake, 2018-06-03:
Maybe this is also related?
[image]

McGiverGim, 2018-06-03:
Strange. The time is not cut in my screen.

Lord-Quake, 2018-06-04:
This is on a different computer. It seems to be placing itself randomly.
[image]

wolkesson, 2018-06-04:
Sorry, I won't have time to look at this in the nearest days. There might be some side effect to my changes. 

The Square behide sticks and white axis are drawn with a side of 2*R, where R = Math.min(width/4 - 8, height/2 - 8). Stick position is in range [-1,1] is scaled with the same R. So it's not a simple scaling error. There could be a problem with the lookup table pitchStickCurve.lookup(), if it doesn't output in the range [-1,1]. Didn't the rccommand change too in #180 and #190? 

But I would say it's more likely a problem with the resize code, because of the time text positioning error. Do you get any console output errors?

I'm unable to reproduce. Can you post a log here?

Lord-Quake, 2018-06-05:
No worries. This is only a cosmetical problem.
Here another screen shot of what I mean when I say the top is cut off.
It looks like a resizing issue. Example:
[image]
[image]

I cannot always reproduce the text positioning error problem. It happens 'randomly'.

McGiverGim, 2018-06-05:
Changing the height of the canvas seems to fix the issue. 
I will check the code to see where this height is calculated.

Lord-Quake, 2018-06-06:
Today it showed up at this position:
[image]

closed 2018-06-06 by mikeller

McGiverGim, 2018-06-06:
@Lord-Quake can you open a new issue with the "time" problem? This issue with the sticks has been fixed and closed, if the other still remains, please open a new issue and I will try to reproduce it and fix.