betaflight/blackbox-log-viewer #196: Debug field graphs display with seemingly random scaling and position

etracer65, 2018-06-03:
When comparing some debug fields to other normal fields in logs it's common that the scaling or position of the debug data seems to be chosen at random.  Take this example:

[image]

In this case the data in debug[0] is exactly the same scale and range as the data in rcCommand[roll].  The graph setup for this display is:

[image]

In fact, even the data elements at the highlighted point are identical:

[image]

But clearly in this case the scaling is different.

Additionally, it seems if there are multiple logs in the file it causes not only the scale to be incorrect but also offsets the debug values:

[image]

I've also observed (but don't have repeatable steps) cases where adding or removing additional fields to the graph causing the debug values to change scaling or position.  In some cases quitting and reopening the same log (with unchanged settings) will also change the display.

McGiverGim, 2018-06-04:
The Debug values must be added, one by one by each debug mode. If not, the blackbox explorer ignores what this debug value contains, so it shows the raw value but ignores the max or min, offset, etc.

To any debug mode, the "names" for the fields can be added in `flightlog_fields_presenter.js` and the scale, etc. in `graph_config.js`.

Take a look to it, or if you prefer, please give here the information for the debug mode (name, name of each debug field, units, value that contains) in an easy way that I can understand, attach one log file, and I will add it ;)

etracer65, 2018-06-04:
I can do that, but perhaps we need a fallback for unknown debug_mode values?  I run into this all the time when testing new code and add a debug_mode to see what's going on.  Being able to able to align the data to other fields (even between debug 0 and debug 1 for example).  In the end these fields may not ever make it into a PR.

etracer65, 2018-06-04:
Or maybe an option in the graph setup to "match" the offset/scaling of some other known field - just an idea.  So if I know that I want to compare debug[0] to rcCommand[roll] I can match the scaling/offset because I know they have the same raw values.

wolkesson, 2018-06-15:
What about creating a general "match function" that you can use to visually match scale of any two fields? Match the min and max value using the full length data of the two fields. Maybe this can also be extended to matching expo? Sure, you may match completely unrelated signals, but I just guess you'll have to use it with care.

mikeller, 2018-08-15:
@etracer65: Marked as 'bug' to keep it open.

closed 2026-08-30 by blckmn