betaflight/blackbox-log-viewer #197: Opening a different log with the same name fails to load

etracer65, 2018-06-04:
I'm sure this is some artifact of the file open handler and this problem has been around forwever, but the scenarios is this:

It's common to log to an SD card and have the file named LOG00001.BFL.  Then open that log in BBE directly from the SD card.  Delete the file from the SD card, eject and go capture another log.  The new log's name is also LOG00001.BFL.  When trying to subsequently open the new log, BBE assumes the same file was chosen (because it's same name/path I'm guessing) and doesn't open the file - keeping the previously loaded data.  Opening another file before or quitting/restarting BBE is a workaround.

It's gotten me a few times while testing because I didn't notice that the new log didn't open and I was looking at the previous.  Maybe we can eliminate this efficiency check and just reopen the log always when the user specifies a file.

McGiverGim, 2018-06-04:
Some questions:
- Are you using the latest master?
- Are you opening the log from the "open" button, double clicking the log file or with drag and drop to the window?

Thanks!

etracer65, 2018-06-04:
Yes, latest master.

Opening the log via the "open" button.  Just tested with drag-drop and that seems okay.  So it's just with the file dialog.  Testing on OS X if that matters.

etracer65, 2018-06-04:
Unrelated, but displaying some kind of "Loading..." status like is happening on the startup window would be helpful (the "Trying to load..." message).  When loading another large log after one is already open the viewer just freezes while it's loading with no indication of what's going on.  Sometimes big logs take a moment to open.

McGiverGim, 2018-06-04:
Yes, that will be useful. And when loading the spectrum analyzer too. This two tasks are slow.

McGiverGim, 2018-06-07:
Sorry, it was the inverse, the live is deprecated. I will try to search the solution, but as I say, is no our code, is the JQuery on change default behaviour.

McGiverGim, 2018-06-07:
Workaround found ;)

PR #208 (2018-06-19): Debugging without nw.js
Allow for debugging most of the application without rebuilding it. This saves a lot of time! It can now be run directly from the file system, with some limitations in video-export. In VS Code for example you may start a debug session with the follow contents in `.vscode/launch.json`:

```
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "chrome",
            "request": "launch",
            "name": "Launch Chrome",
            "url": "${workspaceFolder}/index.html",
            "webRoot": "${workspaceFolder}", 
        },
    ]
}
```
Or, y [... cut]
closed 2018-06-08 by mikeller