betaflight/blackbox-log-viewer #201: Time stamp is positioned randomly near legend.

Lord-Quake, 2018-06-06:
E.g. can occur when window is resized to full screen (full screen button).

See example screen shots.
[image]
[image]
[image]
[image]

Tested on latest builds.

Lord-Quake, 2018-06-07:
OK, I can now replicate the positioning error.
(I open the program by double clicking on a log file)

- Open the program and set it to full screen.
- Close the program. 
- Open the program once again -> The time stamp is positioned correctly.
- Resize the program to e.g. half screen size -> The time stamp is positioned incorrectly.
- Close the program.
- Reopen the program -> The time stamp is positioned correctly.
- Set program to full screen -> The time stamp is positioned incorrectly.

I'm sure the involved programmer can now find the culprit and fix the issue.

McGiverGim, 2018-06-07:
Thanks! It was easier than that:
- Open the program
- Make it half size
- Open a log
- Maximize

Now it is fixed ;)

Lord-Quake, 2018-06-07:
Yes, I know but you devs always are picky and want lots of details ;-)

closed 2018-06-08 by mikeller