betaflight/blackbox-log-viewer #207: Duplicated function getManifestVersion

wolkesson, 2018-06-12:
`getManifestVersion` is duplicated in both main.js:1991 and index.js:56. Logically it belongs more in index.js that does the actual checking for updates, but because of the order the files are included in index.html it need to also be available in main.js. Where does it belong? Maybe in tools.js? Any other or new js-file?

McGiverGim, 2018-06-13:
I suppose that the ideal will be to have a new file only for chrome manifest and other related functions... but I suspect we only have this one. So maybe in the `tools.js` is a good place. There are others "unrelated" functions there, that check for versions (firmware version).

mikeller, 2018-06-13:
Yeah, sorry, that mess is entirely mine. A lot of the GitHub API / version check code was written / copy pasted last year when I was under the pump to get a standalone version of the configurator to the users before Google shut down the Chrome Web Store for non-Chromebooks (which they then didn't do when it had been announced. Later on it got copied into the blackbox log viewer.

Any refactoring is welcome, but it should probably be done in a way that makes it possible to re-use it in the configurator as well? (Should we look into introducing re-usable components and share?)

wolkesson, 2018-06-19:
It would most certain be nice with reusable components between the betaflight projects! I'm also looking int using a framework for components like [Vue](https://vuejs.org/) or Angular. I think we should open a ticket about that to spawn the discussion. This can be closed in the light of #208

closed 2018-06-19 by wolkesson