betaflight/blackbox-log-viewer #213: Mac-OSX legend scrolling

Bradw-17, 2018-06-23:
Scrolling down in the legend does not work, leaving some fields unreachable.

etracer65, 2018-07-03:
I tested with the current release on Mac OS X and the scrolling works fine. Maybe you're referring to using the scroll wheel on your mouse?  That doesn't seem to work but might be a limitation in the application framework.  You can scroll with the scroll bar.

McGiverGim, 2018-07-03:
The scroll with the mouse in the legend is used to change the zoom, expo and smooth of each of the graphs. You can look at the help dialog to see exactly the combinations.

Bradw-17, 2018-07-03:
I'm using the latest Jenkins build, v3.1.0. I have no scroll bar to use, the only way I can access the lower fields is by clicking in a very specific location (circled in red in the screenshot)
[image]

McGiverGim, 2018-07-04:
@Bradw-17 can you export the workspace (top button) and attach it here? what resolution/size are you using?

Bradw-17, 2018-07-04:
Here's the workspace file, and sorry are you referring to my Macbook screen or a setting on the explorer?
[workspaces.json.zip](https://github.com/betaflight/blackbox-log-viewer/files/2164100/workspaces.json.zip)

McGiverGim, 2018-07-04:
Your screen to know more it less the size of the explorer

Bradw-17, 2018-07-04:
1280X800

etracer65, 2018-07-06:
@Bradw-17 Check your System Preferences/General.  Do you have your scroll bar settings like this?

[image]

If so, try changing to one of the other settings and test again.

etracer65, 2018-07-06:
@McGiverGim The problem is when OS X is set to the "When scolling" setting the scroll bars only show up when the scrolling using the mouse wheel or trackpad.  The problem I think is that BBE is capturing the scrolling in the legends pane to adjust smoothing.  Also I thing the "When scrolling" option is the default in OS X so this presents a problem.  I would suggest we disable the scroll wheel smoothing adjust if the target is OS X.  Maybe in general as people seem to be surprised by it anyway.  I'm not sure how useful it is to use the scroll wheel to adjust smoothing as it's kind of a hidden feature.

McGiverGim, 2018-07-06:
It seems there are tons of solutions, for example here are some: https://stackoverflow.com/questions/7855590/preventing-scroll-bars-from-being-hidden-for-macos-trackpad-users-in-webkit-blin

But I don't have a Mac to check which is the best.

mikeller, 2018-07-06:
@McGiverGim: I think the correct solution here is to remove the binding of the scroll wheel to smoothing when in the legends pane. Even in other environments that show scroll bars, I think it is irritating for most users if, when the mouse pointer is in a panel that can scroll, the mouse wheel action is not scrolling. This is a default metaphor (it's called 'scroll wheel' for a reason), and breaking the metaphor isn't good UX.

McGiverGim, 2018-07-06:
Some of the operations that can be done with the scroll are joined with `alt` or `shift` key. Maybe we can try to add a key to the smooth too and if no key pressed then act as normal scroll and don't intercept it. I think this will work. I will look into this, maybe today.

McGiverGim, 2018-07-06:
Done, @etracer65 @ctzsnooze @Bradw-17 can you test it?

Bradw-17, 2018-07-06:
I don't know how to download different builds unless they're up on the Jenkins site unfortunately.

McGiverGim, 2018-07-06:
Don't worry, etracer had tested it. It works but it seems it need some adjust.

closed 2018-07-08 by mikeller