betaflight/blackbox-log-viewer #227: Cannot see everything on low resolution display.

grizewald, 2018-07-09:
A small problem, but a problem none the less.

I use a small laptop with a 12" screen for my multirotor stuff. The screen resolution is only 1366x768 and the Welcome page overflows the screen without presenting a scroll bar to see vaguely important things like the version of the Black box Explorer. (This is the very latest 3.1.0 version which I installed earlier today from the amd64.deb stand-alone package. I'm running Linux Mint 18.3)

Here's what I see:

[image]

Any chance of a scroll bar for the resolution challenged?

McGiverGim, 2018-07-13:
Fixed, thanks for let us know ;)

grizewald, 2018-07-13:
Lovely!
I tested the fix on a local installation and it works just fine. Many thanks!

closed 2018-07-13 by grizewald