betaflight/blackbox-log-viewer #228: Feature Request / WIP: Spectrogram

wolkesson, 2018-07-12:
I'm working on a spectrogram for analysing vibrations (for example)! There's lot of info about it on the net, but a quick explanation would be; vertical is frequency, horizontal is time, color is amplitude. This allows you to, for example, see how the motor output affects the vibration and if the dynamic filter catches it. This would reveal if there are problematic resonance bands or if the one engine gives more vibration than another. 

I have a proof of concept that looks promising, but there are still a lot to do until it's integrated. For example how would you select the field to draw as a spectrogram (you probably just what one) and how to handle multiple graph lanes. It's surprisingly fast, but still need some optimization to run at full playback speed. It's also a question on how much memory we can allow it to use for caching data.  Here's a screenshot, where I've plotted motor 1 against raw roll gyro and the dynamic filter's center frequency : 
[image]

To get the energy of the signal the output of the real part of the fft is squared. But using the non-squared result gives a betting plot. Also giving more weight to high frequencies also gives a better looking plot. However that  will probably fool the user into thinking they have a huge problem with vibrations. 

Just wanted to give a shout in case someone is working on the same thing. It'll take a while until this is a PR, but I'll commit a branch on my fork in a few days.

McGiverGim, 2018-07-13:
It is spectacular, but my knowledge about this is very reduced so at this moment I can't interpret it. I will need to look at some tutorials after you finished that. Thanks for the work!!! Seems very interesting.

McGiverGim, 2018-07-16:
Without knowing how to use it is difficult to say. I suppose that you will always take it from an actual graph, so maybe have an arrow next to each graph that shows a menu can be useful and in a future more options can be added. 
But you know more about the use cases for this.

haslinghuis, 2024-05-29:
@wolkesson closing as stale - now we have a PiP. Like to have it in the main graph though.

closed 2024-05-29 by haslinghuis