betaflight/blackbox-log-viewer #231: Player freeze after changing any graph setting

ghden, 2018-07-15:
I recorder a video to show how it works but after changing something and try to play again, does not play the log.... link to 29 seconds video https://www.youtube.com/watch?v=RF7Q96vc2n4.

Attached one of the logs that makes that behavior..

I switched back to the 3.0 and no problem at all... it does this with at least 4 logs files I have generated with BF 3.4 released..  I didn't test with older BF firmware.... i don't have any now...

Thanks
Denis.
[LOG00004.BFL.zip](https://github.com/betaflight/blackbox-log-viewer/files/2196251/LOG00004.BFL.zip)

McGiverGim, 2018-07-16:
I'm not sure how to reproduce it.
I have downloaded your log, opened it in blackbox, pressed play and the log was reproduced without problem. I need to make something more?

ghden, 2018-07-16:
ok, I only do what I put on the video (I hope you watched it), was the simplest way I found to reproduce consistently, other ways I do is to add or remove more graph parameters, etc. I running the standalone version on Mac OS-X, I installed back the 3.0 and keeping the new version I can open it with 3.0 and no problem... and closing and opening on the 3.1 I have the issues...

ghden, 2018-07-16:
I installed on my mac mini and no problems I will reinstall on macbook pro to see if works....

ghden, 2018-07-17:
I did a clean installation again in the macbook pro and the same issue with this version 3.1, 3.0 works fine.... weird....

McGiverGim, 2018-07-17:
I don't have a Mac. Maybe is a Mac issue only. I don't know if you can test it @wolkesson to see if you can reproduce it.

ghden, 2018-07-17:
Both macs have the same OS version, differences are in graphics and resolutions, i using standalone verion of the app. If I play and do not change anything it does correctly... I will try to do more tests tonight to get you more information... BTW the expo, smooth and grid icons right there is an awesome feature... I would like it more if I can change the % value right there... @wolkesson  if you tell me little more why you thinking on storage problem I can find out in my side.

wolkesson, 2018-07-17:
Unfortunately, I don't have a mac either. :( I couldn't reproduce it on my windows 10 with neither 3.0 nor 3.1. 
What I meant with storage was that at main.js:1556 there's a function called saveOneUserSetting. This saves settings to a file or database somewhere on your computer. If this fails I see no error handling. Maybe it fails because of some invalid file in one of the versions but for the other the file is ok. Does the same thing happen if you press "Overview > Show/hide Stick display" or "Overview > Show/hide Craft display"

Ps. You can change the %-values using scroll and key-combinations over the fields in the legend. See the help under help button top right. Is this what you're looking for? Should work for mac too.

ghden, 2018-07-18:
Well Show/Hide sticks or craft does not affect the behavior it works... is anything I do with the graph, adding/removing, etc.. changing stuff on the graph... I can change speed, zoom and still continue working fine...

Thanks, I didn't read the manual, my bad... yes the scrolling works on Mac...

ghden, 2018-07-18:
I found the blackbox explorer stuff at /Users/myuserid/Library/Application Support/betaflight-blackbox-explorer, there is a lot of stuff there, including default directory that there is a local storage underneath I deleted the whole betaflight blackbox explorer folder and opening the application again, recreate everything with no problem and still the exact same issue when I modify anything on the graph.. I have several GB of free space on HD...  I don't see errors in the MAC that works... and I see the following in the one does not work (attached file)
[system logs.txt](https://github.com/betaflight/blackbox-log-viewer/files/2206829/system.logs.txt)

Another thing is it just stop playing the log, but if I click on the play/pause icon again, it toggles from pause to play, but still does not move, but the app is still working and responsive at this point everything works, except anything on the playback... I hope this help..

McGiverGim, 2018-07-19:
The only thing that maybe can help, is to use a debug build, and see if there are some errors in the console. But I can't build it because I don't have a Mac. Are you able to build the app by yourself?

ghden, 2018-07-19:
I haven't but I can do it... Granted Im not a java guy but I should be able to build it...

McGiverGim, 2018-07-20:
Perfect, if you finally are able, start the program with `gulp debug` and you will be able to press the mouse right button and open the developers tools, and see if some error appears in the console that can point us in the right direction.

salem83, 2018-07-26:
I got the same problem on windows. 
The graphs (top an bottom) will disappear if i go past the middle of the logs

[image]

Doesn't matter which log i use. its the same issue if the log is 25Kb or 7MB.