betaflight/blackbox-log-viewer #245: Analyser not working for a BB logs from BF 3.5RC2

supiiik, 2018-08-12:
Hi,
analyser is not working in 3.2.0 for a data from BF 3.5RC2. Data from version BF3.4 are working properly.
I can see only blank analyser window there
[image]

mikeller, 2018-08-12:
Testing with a 3.5.0-RC2 log file on Blackbox Log Viewer 3.2.0, works fine for me. Can you please post a log file that does not work?

supiiik, 2018-08-12:
[LOG00022.ZIP](https://github.com/betaflight/blackbox-log-viewer/files/2280836/LOG00022.ZIP)

mikeller, 2018-08-12:
@supiiik: Your log has got some missing frames, and one of them is triggering a bug when trying to open the analyser. It is fixed in #246. If you let me know what platform you are on I can build you a version to test.

supiiik, 2018-08-12:
I'm on Windows. Thanks

mikeller, 2018-08-12:
Here you go:
https://drive.google.com/file/d/1OAi5FE3JBXQA5qiUYyZUAc9YbqKr81AQ/view?usp=sharing

supiiik, 2018-08-12:
It's working now. Thanks :D

PR #246 (2018-09-21): Fixed bug when parsing invalid 'E' frame.
Fixes #245.
closed 2018-09-18 by stale[bot]