betaflight/blackbox-log-viewer #247: 3.2 MacOS binary broken

ljfh001, 2018-08-12:
Please see attached:
[image]

etracer65, 2018-08-12:
Can you provide more details like your OS version?

I just tested with OS X versions 10.10, 10.12 and 10.13 and all worked just fine.  Possibly just download it again or maybe use Safari to download instead.

ljfh001, 2018-08-12:
10.13.6. Tried downloading it again. Still the same error.
It works perfectly fine when I launch from CMD:
/Applications/Betaflight\ Blackbox\ Explorer.app/Contents/MacOS/nwjs

ljfh001, 2018-08-12:
sudo spctl --master-disable worked
I could do option right click to open it before. weird...

yorb, 2018-08-15:
Same issue here, macOS 10.13.6. Tried with Chrome 68.0.3440.106 (current) and Safari 11.1.2 (13605.3.8) (also current).

This is definitely a new issue with 3.2.0, never seen it before. Let me know how else I can help troubleshoot!

closed 2018-09-22 by stale[bot]