betaflight/blackbox-log-viewer #258: Debug data GYRO_SCALED displays normal gyro data instead

etracer65, 2018-11-20:
When data is logged using `debug_mode = GYRO_SCALED` the data output is the raw scaled gyro data before any filtering. However when displayed in the log viewer the data presented is identical to the normal gyro data. Testing with an old version of the log viewer doesn't have this problem and the correct data is displayed. It looks like the `GYRO_SCALED` data might be being erroneously considered a calculated field based of the gyro values rather than simply presenting it as a debug value like it is.

Here's a short log for testing:
[LOG00002.BFL.zip](https://github.com/betaflight/blackbox-log-viewer/files/2600118/LOG00002.BFL.zip)

McGiverGim, 2018-11-20:
As far as I know, all the gyros datas are presented as is, without any kind of conversion and much less any filtering. The only conversion that the code does is to transform it from raw to degrees per second, but nothing more...
As you say, I see the data too "smooth" to be a raw gyro. Are you sure that an older version of log viewer showed the data different?

McGiverGim, 2018-11-20:
I have searched in my BB files, and something must be wrong with yours, in mine the gyro_scaled is shown correctly:
[image]

So it seems the problem is in the data of your file, not in the viewer.

etracer65, 2018-11-20:
These parts of the code had me concerned:

https://github.com/betaflight/blackbox-log-viewer/blob/b38379618ae2104615bcfdc7f20a9db082893fdf/js/flightlog_fields_presenter.js#L59-L63

Referring to a "virtual field".

https://github.com/betaflight/blackbox-log-viewer/blob/b38379618ae2104615bcfdc7f20a9db082893fdf/js/flightlog.js#L14

"GYROADC_SCALED"

https://github.com/betaflight/blackbox-log-viewer/blob/b38379618ae2104615bcfdc7f20a9db082893fdf/js/flightlog.js#L603-L608

I'm wondering if there was already a "scaled gyro" concept and this got confused with the debug_mode "gyro_scaled".

McGiverGim, 2018-11-20:
The GYROADC_SCALED is different from the GYRO_SCALED debug value. I'm not too sure what shows the first, but the second is the raw value as I show you in the capture. I will try to check both tomorrow, but maybe we are talking of different things.

etracer65, 2018-11-20:
Here's the graph config used on the current version.  Note that the debug value is correctly recognized as "Gyro Scaled" so that's good.

[image]

McGiverGim, 2018-11-20:
It can be the wrong gyro scaled, there are two:

The ADC is this:
[image]

The good is the one under the debug:
[image]

I think I need to rename the first adding the ADC to differentiate one from the other ;)

etracer65, 2018-11-20:
Yes, that's it! What is the "Gyro Scaled" ADC anyway? It doesn't seem to be any different from the normal gyro traces. Maybe it's no longer relevant?

McGiverGim, 2018-11-20:
Maybe. I didn't remove it because I don't understand what is. I will open a PR this week renaming it to discuss if better rename or remove it. Maybe someone knows what is for 😋