betaflight/blackbox-log-viewer #260: AirMode appears incorrectly as "Servo2" in the log viewer on the latest 4.0 hexes.

gvhlaw, 2018-12-01:
I've noticed on every log I have flown on the later builds of 4.0 and the variant branches that I am testing that when I switch from Angle/Horizon into Air Mode, what I see in the log viewer is not Air Mode but "Servo 2."
[image]

McGiverGim, 2018-12-03:
The only change in the "modes list" was made here if I'm not wrong:

https://github.com/betaflight/betaflight/commit/017bdca593cd6134665df1403d0c9963e677cbde#diff-c22338cc34442857de77b51a57b0297a

@mikeller looking at the code, it seems to me that the objective was to remove the mode, but let the same ID number for it, so in theory no blackbox change is needed. Am I right? If true, then something maybe is wrong in the firmware part when writing to the blackbox file. If not, then I can modify the blackbox to accept the new IDs (the strange part is that the new number assigned to the mode does not link with the number of elements removed, so maybe the bug is in the firmware if other part of code).

mikeller, 2018-12-03:
Gah, does blackbox use the enum indices, and not `permanentId`? That's a bit unfortunate, but I think we should just handle this in the log viewer in this case, and change the mapping there based on the version in the log.

closed 2018-12-19 by mikeller

gvhlaw, 2018-12-20:
Thanks Mike!  I'll flash the newest Master to get the fix.

-- 

*Confidentiality Notice*:  This e-mail transmission and any accompanying
document(s), may contain confidential information which is legally
privileged under the attorney-client or attorney work product
privileges. Forwarding or sending attorney-client emails or attachments
whether directly or in an email string to persons other than the attorney
and client engaged in an attorney-client relationship may limit or destroy
the attorney-client privilege.  The information contained herein or any
attachment hereto is intended only for the use by the individual(s) or
entities named as the addressee(s) above.  If you are not the intended
recipient, or the person responsible for delivering it to the intended
recipient, you are hereby notified that any disclosure, copying,
distribution, or use of any of the information contained in this
transmission is strictly *PROHIBITED*.  Any such inadvertent disclosure of
information, which includes any metadata in emails or attachments, shall
not compromise or waive the attorney-client privilege as to this
communication or otherwise.  If you have received this transmission in
error, please notify me by telephone at (209)-956-9940 or reply e-mail and
delete this message and any attachments.  Thank you.

George V. Hartmann, Esq.
THE HARTMANN LAW FIRM
3425 Brookside Road, Suite A
Stockton, CA 95219

209.956.9940 O
209.956.9929 F

Discourage litigation. Persuade your neighbors to compromise whenever you
can. As a peacemaker the lawyer has superior opportunity of being a good
man. There will still be business enough.

Abraham Lincoln
16th president of US (1809 - 1865)