betaflight/blackbox-log-viewer #270: Header page scroll broken

McGiverGim, 2019-01-28:
Since the update of NW to the latest version here: https://github.com/betaflight/blackbox-log-viewer/pull/264 the scroll of the Header page is broken.

I have tried to update it to the most newest version without luck.

I don't know if there will be a hack in code to fix it or we will need to go back until fixed in NW before the release.

McGiverGim, 2019-01-28:
Seems there is a similar open issue in NW: https://github.com/nwjs/nw.js/issues/6920

McGiverGim, 2019-01-28:
Tested versions 0.35.5 and 0.36.0-beta1 without luck. Resizing the Blackbox Viewer window makes the scroll work again.

mikeller, 2019-01-29:
Maybe we should just roll back, and keep watching the nwjs issue, and update if there is a new version with a fix long enough before the next release?

McGiverGim, 2019-01-29:
At least we must do it before the release of next version of the Blackbox Viewer. I'm out of the computer now, but I can do it this afternoon.

mikeller, 2019-01-29:
We can as well do it now - if the version we are rolling back to has bugs affecting us as well then I'd rather roll back now, and have a chance of discovering them, than rolling back 'last minute'.

McGiverGim, 2019-04-03:
Tested the newest versions:
- Latest stable 0.37.2: The problem persists.
- Latest beta 0.38.0-beta1: The problem seems fixed. When the final version will be released we can update the version in our side too.

mikeller, 2019-04-04:
Thanks for testing, @McGiverGim. I guess we can update in time for 3.4.0.

closed 2019-04-30 by mikeller