betaflight/blackbox-log-viewer #282: Betaflight Blackbox Explorer crashes

jasenhowell, 2019-03-14:
Betaflight Blackbox Explorer crashes after 2 seconds of playing  log or moving along the quick access bar. any ideas?

mikeller, 2019-03-15:
@jasenhowell: I have got an idea: How about you supply a bit more information, like what version of the Blackbox Log Viewer you are using, and the log file / video that you have opened when this is happening?

jasenhowell, 2019-03-15:
here is the link to the video   
https://youtu.be/HartWlsjCys

jasenhowell, 2019-03-15:
[https://1drv.ms/u/s!AsgLZPDv54KstHR0zWtulDtIr65-](url)

jasenhowell, 2019-03-15:
version 3.2.0

mikeller, 2019-03-16:
I can reproduce the problem in 3.2.0, and it looks like it's the one that was fixed in #246. As a confirmation, the problem is fixed in the latest version from `master`, which will be released as 3.3.0 within a few weeks.

On a side note, your log has a lot of dropouts. You might want to look into reducing the logging rate.

closed 2019-04-22 by stale[bot]