betaflight/blackbox-log-viewer #293: export workspaces fails / debian linux

meoso, 2019-03-30:
v 3.3.0 (& 3.2.0)

Export workspaces saves .json with all null values.

```
{
    "graphConfig": [
        null,
        null,
        null,
        null,
        null,
        null,
        null,
        null,
        null,
        null
    ]
}
```

Docteh, 2019-04-10:
Interestingly enough I get that on windows as well.

McGiverGim, 2019-04-10:
I have tested it right now with Ubuntu and is working for me.

What maybe has happened, is that I think you have not assigned the graphs to a favorite. You must assign it pressing `shift + number` (for example `shift + 1`) and then export the workspace. In this way you can recall the graphs pressing `1`.

Only favorites are assigned to the workspace.

meoso, 2019-04-10:
@McGiverGim you have saved me a thousand hours of undesired tinkering! THANK YOU!!!

closed 2019-04-10 by meoso