betaflight/blackbox-log-viewer #300: Log file selection list missing. (& typo)

NinjaSauce1, 2019-04-11:
I haven't used the viewer in a while, and when I went to use it today, I can't seem to locate the log file list selection drop-down menu. There seems to be no way to access the different logs. I have looked everywhere and I cannot seem to find it. I definitely have multiple logs on this file. Is this a glitch or is it just hidden now? I will attach a screenshot.

Also, there is a typo on the pop-up when hovering over the "Export Video" button.
It reads "Export your vide" and I think it might be supposed to say Export your video....
It will also be shown in the screenshot.

[image]

McGiverGim, 2019-04-11:
The different logs are in the combo, on the right, just under the Legend text. I will fix the typo, thanks!

NinjaSauce1, 2019-04-11:
Slaps forehead. Thanks!

closed 2019-04-14 by mikeller