betaflight/blackbox-log-viewer #310: Blackbox log viewer 3.3.0 - Log files on firmware betaflight 3.5.x are displaying strange related gyro datas graphs

blackmennewstyle, 2019-04-22:
Is it normal when i try to open log files on firmware 3.5.x with the Betaflight Blackbox Viewer 3.3.0 that i get strange graphs for most gyro datas?

- Log file with firmware Betaflight 3.5.5
[image]

- Log file with firmware Betaflight 3.5.7
[image]

Do i have to run the dedicated blackbox viewer for each different version of betaflight firmware i am using on my panel of quads? :'(

McGiverGim, 2019-04-22:
This has been fixed in master: https://github.com/betaflight/blackbox-log-viewer/pull/308 until a new release, you can build yourself a version or go to your installed 3.3.0 directory, all the files are there in plain text, and make the same change of the PR.

blackmennewstyle, 2019-04-23:
> 
> 
> This has been fixed in master: #308 until a new release, you can build yourself a version or go to your installed 3.3.0 directory, all the files are there in plain text, and make the same change of the PR.
> 
> EDIT: Sorry, only the Configurator is in plain text, Blackbox are compressed.

Thank you so much for the answer :)

No worry i installed the 3.2.0 blackbox viewer in a virtual machine so i can deal with both. I won't be using them both for a long time anyway, currently moving everything to Betaflight 4.0.0 <3

Have a great week and keep on flying!

McGiverGim, 2019-04-29:
@blackmennewstyle a new release 3.3.1 has been published that must fix it. Can you test it please?

blackmennewstyle, 2019-04-29:
> 
> 
> @blackmennewstyle a new release 3.3.1 has been published that must fix it. Can you test it please?

Hi :)

Thanks for the heads-up. I just tested it out and everything look to be fixed and working properly :D

- Log file with firmware Betaflight 3.5.5
[image]

- Log file with firmware Betaflight 3.5.7
[image]

Have a great week :)

mikeller, 2019-04-29:
@blackmennewstyle: In this case can you close the issue please?

blackmennewstyle, 2019-04-29:
> 
> 
> @blackmennewstyle: In this case can you close the issue please?

Oh sorry. Done :)

closed 2019-04-29 by blackmennewstyle