betaflight/blackbox-log-viewer #320: RPM install uses wrong path for .desktop file

JoveToo, 2019-04-30:
>  Running transaction
>   Preparing        :                                                        1/1 
>   Upgrading        : betaflight-blackbox-explorer-3.3.1-1.x86_64            1/2 
>   Running scriptlet: betaflight-blackbox-explorer-3.3.1-1.x86_64            1/2 
> cp: cannot stat '/opt/betaflight/betaflight-blackbox-explorer/mime/betaflight-blackbox-explorer.xml': No such file or directory
> cp: cannot stat '/opt/betaflight/betaflight-blackbox-explorer/icon/bf_icon_128.png': No such file or directory
> gtk-update-icon-cache: Cache file created successfully.
> xdg-desktop-menu: file '/opt/betaflight/betaflight-blackbox-explorer/blackbox-log-viewer.desktop' does not exist
> warning: %post(betaflight-blackbox-explorer-3.3.1-1.x86_64) scriptlet failed, exit status 2
> Non-fatal POSTIN scriptlet failure in rpm package betaflight-blackbox-explorer
> Non-fatal POSTIN scriptlet failure in rpm package betaflight-blackbox-explorer
>   Running scriptlet: betaflight-blackbox-explorer-3.3.0-1.x86_64            2/2 
> rm: cannot remove '/usr/share/mime/packages/betaflight-blackbox-explorer.xml': No such file or directory
> rm: cannot remove '/usr/share/icons/hicolor/128x128/mimetypes/application-x-blackboxlog.png': No such file or directory
> gtk-update-icon-cache: Cache file created successfully.
>   Cleanup          : betaflight-blackbox-explorer-3.3.0-1.x86_64            2/2 
>   Verifying        : betaflight-blackbox-explorer-3.3.1-1.x86_64            1/2 
>   Verifying        : betaflight-blackbox-explorer-3.3.0-1.x86_64            2/2 
> 

But installer installs in /opt/betaflight/blackbox-log-viewer not /opt/betaflight/betaflight-blackbox-explorer

mikeller, 2019-04-30:
@JoveToo: Can you try this to see if it is better:

https://drive.google.com/open?id=1koHB8_I1Zr7mfZ1kZSonIBUNGVA9gci5

JoveToo, 2019-04-30:
Hi,
yes now it installs in /opt/betaflight/betaflight-blackbox-explorer

But I didn't find the icon in my menu I had to execute:

> $ xdg-desktop-menu install /opt/betaflight/betaflight-blackbox-explorer/betaflight-blackbox-explorer.desktop 
> $ xdg-desktop-menu forceupdate

mikeller, 2019-04-30:
@JoveToo: Try this: https://drive.google.com/open?id=1koHB8_I1Zr7mfZ1kZSonIBUNGVA9gci5

JoveToo, 2019-05-01:
@mikeller It seems to work and the icon is in my menu, thank you.

PR #321 (2019-05-01): Fixed invalid pathing in RPM build.
Fixes #320.
closed 2019-05-01 by mikeller