betaflight/blackbox-log-viewer #329: No lines in graph

flatwound, 2019-05-05:
When opening the bbl from a bf 4.0 quad, I can see there's data, but the lines are not plotted.

using the linux64 bit archive.

[image]

McGiverGim, 2019-05-05:
Attach your log please.

flatwound, 2019-05-05:
Done

McGiverGim, 2019-05-06:
At least in windows, it works:
[image]

I will try to test it under Linux later.

McGiverGim, 2019-05-06:
Tested under linux:
- The first time I openend the BBL file, the same than you. No lines.
- Modified the graphs adding one, pressing the `Graph setup` button, and all ok.
- Now all is ok every time I open a BBL file.

Maybe some old data of older versions, that is refreshed in the Graph setup. Can you test to add some graphs?

flatwound, 2019-05-06:
Yep works when adding additional graph.. thx!

jhillyerd, 2019-09-15:
I've never installed or run the black box explorer before today, and ran into this issue.  v3.3.1 win32 on Win 10

junwoo091400, 2019-10-29:
Maybe this is a fixable issue? ;(

SantiCO19, 2020-01-30:
Same issue here on Win10, BBLV version 3.4.0 win32

Christian-Seematter, 2020-05-13:
Same issue here on Win 10, version 3.5.0 from betaflight-blackbox-explorer-installer_3.5.0_win32.exe

dkijkuit, 2020-06-23:
Same issue here on a fresh Win10 system. After pressing 'Graph setup' and 'Save changes' the graph suddenly appeared.

McGiverGim, 2020-06-23:
We usually associated this to an update of the blackbox. Your windows 10 never had blackbox installed before?

dkijkuit, 2020-06-23:
> We usually associated this to an update of the blackbox. Your windows 10 never had blackbox installed before?

No I actually used the blackbox for the first time ever ;)

sn0opy, 2020-07-03:
I can confirm @dkijkuit's issues. Same here. First time opening on Windows and had to follow the steps described above to get my graphs going.

McGiverGim, 2020-07-03:
I reopen this issue. Needs to be investigated.

ouinouin, 2020-09-17:
same issue here under linux and V3.5

closed 2026-08-30 by blckmn