betaflight/blackbox-log-viewer #333: VTXPITMODE changes when disabling and enabling ANGLE mode

IvoFPV, 2019-05-06:
I was just testing my build today hovering with angle mode and disabled and enabled it a couple of times and after that I checked blackbox logs and it shows that every time when I did that, VTXPITMODE was also disabled or enabled. Pit mode is not assigned in modes tab. Running BF 3.5.5 on this quad, using latest log viewer 3.3.1

[image]

[LOG00023.zip](https://github.com/betaflight/blackbox-log-viewer/files/3148434/LOG00023.zip)

# diff

# version
# Betaflight / MATEKF405 (MKF4) 3.5.5 Jan 15 2019 / 10:05:36 (88921670b) MSP API: 1.40

board_name MATEKF405
manufacturer_id 

# name
name Ivo

# resources
resource MOTOR 5 NONE
resource SERIAL_TX 11 A15

# mixer

# servo

# servo mix

# feature
feature AIRMODE

# beeper

# beacon

# map
map TAER1234

# serial
serial 4 2048 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1200 2100 0 0
aux 1 1 0 1200 1600 0 0
aux 2 13 1 1800 2100 0 0
aux 3 26 0 1200 2100 0 0
aux 4 35 1 1300 1700 0 0

# adjrange

# rxrange
rxrange 0 987 2011
rxrange 1 987 2011
rxrange 2 987 2011
rxrange 3 987 2011

# vtx

# rxfail

# master
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 8
set rc_smoothing_type = FILTER
set blackbox_p_ratio = 64
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1300
set vbat_min_cell_voltage = 34
set vbat_warning_cell_voltage = 36
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_angle = 35
set gps_rescue_initial_alt = 25
set gps_rescue_descent_dist = 80
set gps_rescue_ground_speed = 1000
set gps_rescue_sanity_checks = RESCUE_SANITY_OFF
set gps_rescue_min_sats = 0
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 50
set osd_cap_alarm = 1000
set osd_vbat_pos = 2466
set osd_rssi_pos = 2381
set osd_tim_1_pos = 471
set osd_tim_2_pos = 2519
set osd_flymode_pos = 2145
set osd_throttle_pos = 2487
set osd_vtx_channel_pos = 2168
set osd_crosshairs_pos = 2285
set osd_ah_sbar_pos = 238
set osd_ah_pos = 2158
set osd_current_pos = 2475
set osd_mah_drawn_pos = 2508
set osd_gps_speed_pos = 227
set osd_gps_sats_pos = 153
set osd_home_dir_pos = 108
set osd_home_dist_pos = 110
set osd_altitude_pos = 246
set osd_pidrate_profile_pos = 129
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2498
set osd_battery_usage_pos = 40
set osd_disarmed_pos = 2443
set osd_core_temp_pos = 2202
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_bbox = OFF
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 2
set vtx_freq = 5695
set rcdevice_init_dev_attempts = 4

# profile
profile 0

set dterm_notch_cutoff = 0
set anti_gravity_gain = 7000
set feedforward_transition = 15
set iterm_relax = RP
set p_pitch = 47
set i_pitch = 55
set d_pitch = 34
set f_pitch = 100
set p_roll = 47
set i_roll = 55
set d_roll = 31
set f_roll = 100
set p_yaw = 40
set i_yaw = 120
set f_yaw = 100

# rateprofile
rateprofile 0

set roll_rc_rate = 192
set pitch_rc_rate = 192
set yaw_rc_rate = 192
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 20
set roll_srate = 64
set pitch_srate = 64
set yaw_srate = 50

#

McGiverGim, 2019-05-08:
It seems that the info of mode changes is stored in a 32 bit variable, as flags. But the list of modes are bigger than 32, so it is overflowing and detecting modes that does not exist.
I think this first needs to be fixed in the firmware part, and then read it accordingly here.

chupocro, 2019-09-09:
Similar here. When I switch from angle to horizon mode the log viewer shows: Flight Mode Change | ANGLE OFF | HORIZON ON | VTXPITMODE OFF | USER1 ON. When switching from horizon to acro mode it says: Glight Mode Change | HORIZON OFF | USER1 OFF. Going back to horizon mode it says: Flight Mode Change | HORIZON ON | USER1 ON and when returning back to angle mode it says: Flight Mode Change | ANGLE ON | HORIZON OFF | VTXPITMODE ON | USER1 OFF.

I don't have assigned anything for VTX Pit Mode or USER1. Running Betaflight 3.5.0 and log viewer 3.3.1

McGiverGim, 2019-09-09:
The flight modes over 32 give this strange behavior, they are not supported.

closed 2026-08-30 by blckmn