betaflight/blackbox-log-viewer #344: Error while loading the video. ERROR (4): DEMUXER_ERROR_COULD_NOT_OPEN: FFmpegDemuxer: open context failed

striker890, 2019-05-25:
I tried it on two devices and two os (Manjaro/Windows). On both I get this error when trying to open dvr footage of my fatshark hdo's:

Error while loading the video. ERROR (4): DEMUXER_ERROR_COULD_NOT_OPEN: FFmpegDemuxer: open context failed

I tried two recordings and none works. Opening in media players works without problems.
Isn't the video format supported?

McGiverGim, 2019-05-25:
That is what it says. Can you attach a little video to see if I can find some workaround?

IgnazNepumuk, 2019-05-26:
same here on linux64 with fatshark hd3 dvr footage

striker890, 2019-05-28:
Thanks @McGiverGim 
Here's a little demo file. It's just some noise recorded, but it shouldn't matter.
[PICT0003.AVI.tar.gz](https://github.com/betaflight/blackbox-log-viewer/files/3227564/PICT0003.AVI.tar.gz)

McGiverGim, 2019-05-28:
I have been playing with that without luck :( Updating the nw.js and ffmpeg libraries that we use to the latest version, did not fix the issue. It seems that they are not supported by Chromium, the source of nw.js.
I'm not an expert in codecs, but I think we can't do nothing against this.

mikeller, 2019-05-29:
@McGiverGim: I concur. The whole codec situation is reasonably shitty - with licensing issues and rights management concerns playing into this. Video overlay is not a main function of the blackbox explorer, so I think it is sufficient to support the limited number of formats that we do now with the limited effort we have to put into it.
If users are really intent on creating overlays for codecs that we do not support there is always the way for them to convert the footage into a supported format, and then adding the overlay.

closed 2019-07-05 by stale[bot]

MiniGod, 2020-08-12:
> That is what it says.

What is it saying? He obviously asked because he didn't understand the error, and neither do I. What does `open context failed` mean?

I managed to get the video loading by first converting it to an `mp4` file. I used `ffmpeg -i PICT0003.AVI PICT0003.mp4`.

I didn't dig into what it is with the original file that causes this error, whether it's the codec, color-space or something else. The original file uses the `mjpeg` codec, while the mp4 file uses `h264`.