betaflight/blackbox-log-viewer #349: decoded values into CSV file

farmingmachine, 2019-06-26:
Hey,

it would be rlly if we could also export the decoded values (e.g. pitch angle) to the CSV-file instead of only the raw data. Could that be made possible?

Kind regards,

Malik

mikeller, 2019-06-26:
Not sure this makes sense and does not just add bloat - these can easily be calculated based on the raw data.

farmingmachine, 2019-06-27:
Yeah you're right , I just realized that as well. Sorry for the trouble ^^

farmingmachine, 2019-06-27:
> Not sure this makes sense and does not just add bloat - these can easily be calculated based on the raw data.

Could you tell me the factors to calculate the raw pitch and roll into angles values? Since the values are rounded in the viewer, it's kinda tricky to get the real factors

Mahdi-Hosseinali, 2019-07-03:
Is there any reference for decoding the raw values?
I'm specifically interested in motors. Not sure what are the units of the raw data and if the decoded values are percentages of the maximum RPM, duty cycle, etc?

closed 2019-08-09 by stale[bot]