betaflight/blackbox-log-viewer #352: beta flight black box 3.3.1

S-T-A, 2019-07-06:
blackbox 3.3.1 after install on OSX crash**Describe the bug**
after first install into applications folder and clicking on it this appears “Betaflight Blackbox Explorer” is damaged and can’t be opened. You should move it to the Trash.

delated and downloaded again same thing
repeated behaviour is the same

etracer65, 2019-07-06:
Just tested and worked fine. Try a different browser or disable any antivirus software that might be interfering.

closed 2019-07-06 by etracer65

S-T-A, 2019-08-03:
still the same thing i get this pop up notice when i try and open blackbox after it installs.
 “Betaflight Blackbox Explorer” is damaged and can’t be opened. You should move it to the Trash."

S-T-A, 2019-08-03:
When i try to open the black box app a window pops up and i get the warning "        "  I click on the question mark in the lower left side of window. thanking me to this window "The app has been modified or damaged
Im on a Mac OSX 10.13.6 I would greatly appreciate any help the older version worked fine  
The app has been modified, and its code does not match the original signed code. The app may be broken or corrupted, or it may have been tampered with. If you think that an app has been damaged or tampered with, you should delete it by dragging it to the Trash.

Some apps and tools, such as AppleScript or JavaScript applications and some legacy tools, modify themselves after signing. These types of apps cannot be opened unless you override the security settings on your Mac. For more information, see Open an app by overriding security settings." when I tried pressing the control Key and opening the app again it was still the same thing!

etracer65, 2019-08-03:
Just tested (again) on MacOS 10.13.6 and it worked perfectly.

1. Download the current release disk image: https://github.com/betaflight/blackbox-log-viewer/releases/download/3.3.1/betaflight-blackbox-explorer_3.3.1_macOS.dmg
2. Double-click on the .dmg file to open the archive.
3. Copy the Betaflight Blackbox Explorer app to your Applications folder - don't try to open the application directly from the disk image archive.

When you try to open from the Applications filter if you get:

[image]
This is because the application is not signed with an "official" (paid) developer signature issued by Apple. You have to go to the System Preferences -> Security & Privacy and override the security settings ("Open Anyway").

[image]

Then you'll get one last alert - just click "Open".

[image]

For errors about a "corrupted" or "damaged" file - then you have some problem with how you are downloading the disk image. Try a different browser. Or possibly you have some anti-virus, file scanning, or download manager software installed that's damaging the file.

There is nothing wrong with the application - you have some issue on your computer.

S-T-A, 2019-08-04:
like i said before 
[image]

S-T-A, 2019-08-04:
i know how to install the app its not my first time it's my first time running into this and non of what you guys said helps and i don't have a virus or a problem with anything else so thank you for nothing "MIKELLER" thanks for kicking me when i'm asking for help, you're a true gentleman and a scholar

S-T-A, 2019-08-04:
I'm tring to figure it out on my own thank you Etracer65 it doesn't matter what browser i use the issue is the same the signature that is signed doesn't match the what i downloaded its as simple as that i'm working on figurung it out i'll let you know what happens next

etracer65, 2019-08-04:
You appear to be the only person with this problem. Every other MacOS user (including myself) have no problems. So the logical conclusion is that you have some problem with your computer or the process you're using to download/install the application. As already mentioned, opening duplicate issues for the same problem will just result in them being closed.

I suggest you seek technical support for your computer or try downloading and installing the app on another computer.

S-T-A, 2019-08-04:
it works fine on my tablet running windows 10 64bit no problem but on my mac its the same thing nothing has changed i just find it odd when everything else works on my MAC except this

mikeller, 2019-08-04:
@S-T-A: Let me clarify a few things here. The software you are using here is available to you for **free**, made by volunteers in their spare time. The support you are getting for **free** is also provided by volunteers in their spare time. 

So if you think it is appropriate for you to act like you are entitled to have this problem solved (see 'non(sic) of what you guys said helps and i don't have a virus or a problem with anything else'), or if you think it is appropriate for you to attack me for laying out the rules that we have in here to reduce the amount of time that the volunteers answering questions have to waste on duplicate questions, then you are way out of line, and will only get banned from here.

If you want to have this attitude of entitlement, then please stop using this software, switch to using commercial software, where the money you pay for it means you actually are entitled to some form of support.

Just in case you don't understand what 'duplicate' means: It means that the issue in question is describing exactly the same problem that is already described in another issue.