betaflight/blackbox-log-viewer #353: Debug_Mode=DUAL_Gyro_Raw Reversed

spatzengr, 2019-07-11:
It appears like the Debug mode descriptions for Dual_Gyro_Raw are revered for Pitch vs. Roll.

[image]

Docteh, 2019-07-12:
Isn't it a best practice to mount the second gyro 90 degrees different from the first?

Or is raw supposed to be after factoring that in?

etracer65, 2019-07-28:
`RAW` data from the sensors is not translated based on mounting orientation. Roll is the gyro's X axis, pitch is Y axis and yaw is Z axis. So if the two gyros are mounted with a 90 degree offset then it will result in the data you're seeing.

Anyway you'd probably be better off using the `DUAL_GYRO_SCALED` debug mode as that has the roll/pitch values scaled to degrees/sec and aligned based on the sensor orientation.

closed 2019-09-03 by stale[bot]