betaflight/blackbox-log-viewer #356: doesn't open after install :(

S-T-A, 2019-08-03:
When i try to open the black box app a window pops up and i get the warning "Betaflight Blackbox Explorer” is damaged and can’t be opened. You should move it to the Trash." I click on the question mark in the lower left side of window. taking me to this window.

"The app has been modified or damaged"
The app has been modified, and its code does not match the original signed code. The app may be broken or corrupted, or it may have been tampered with. If you think that an app has been damaged or tampered with, you should delete it by dragging it to the Trash.

Some apps and tools, such as AppleScript or JavaScript applications and some legacy tools, modify themselves after signing. These types of apps cannot be opened unless you override the security settings on your Mac. For more information, see Open an app by overriding security settings."

When I tried pressing the control Key and opening the app again it was still the same thing!
Im on a Mac OSX 10.13.6 using Google Chrome
Version 75.0.3770.100 (Official Build) (64-bit) to open the app i assume
I would greatly appreciate any help the older version worked fine

mikeller, 2019-08-03:
@S-T-A: Why are you opening an issue that is for the same issue that you have opened in #352, considering that you were given an answer there? Please stop this.

Duplicate of #352, closing.

closed 2019-08-03 by mikeller

S-T-A, 2019-08-04:
its not a duplicate its real it doesn't work do you honesly really think i'm doing this for fun perhaps you should wait and ask me before you deside to close it as for you black box i gave you the answer you where looking for the signed signature doesn't match and my MAC WON'T OPEN IT SO THERE STILL IS A PROBLEM AND WHAT HAVE YOU DONE??????

mikeller, 2019-08-04:
@S-T-A: See https://github.com/betaflight/blackbox-log-viewer/issues/352#issuecomment-518036354