betaflight/blackbox-log-viewer #357: Blackbox Log Viewer 3.3.1 MacOS : error opening log

JimB40, 2019-08-03:
[image]

Got this message every time I want to open log from my R349.
BF firmware is 3.5.1
Tried different logging rates 500, 1000, 2000.
Installed 3.2 version -> same error
Log file in enclosure.

[BTFL_BLACKBOX_LOG_R349HD_20190803_143416.bbl.zip](https://github.com/betaflight/blackbox-log-viewer/files/3463869/BTFL_BLACKBOX_LOG_R349HD_20190803_143416.bbl.zip)

etracer65, 2019-08-03:
The log file is valid - it just doesn't contain any data. Did you remember to set up the Blackbox mode and link it to an aux channel on the Modes tab? The mode must be active otherwise a log won't be captured.

JimB40, 2019-08-03:
Yes I did. CH8 (AUX4) That's why I have this file. 
If switch is off it wont produce any file on internal dataflash.
Tried to keep "recording" longer with BlackBox mode on. It produced slightly bigger file with  those question marks after "end of log". Don't know if this data format or just garbage.
Any other settings in CLI to set???  

[image]
[BTFL_BLACKBOX_LOG_R349HD_20190803_155857.bbl.zip](https://github.com/betaflight/blackbox-log-viewer/files/3463909/BTFL_BLACKBOX_LOG_R349HD_20190803_155857.bbl.zip)
[image]

etracer65, 2019-08-03:
Provide a `diff` of your settings. Without information about the flight controller we can't even tell if you're using on-board flash or SD card or if you have a misconfiguration. Make sure to erase the device before logging. Also if transferring from on-board flash then use mass storage mode rather than downloading from the configurator. Also you're running a very old version of the firmware (3.5.1). The last 3.5 version release is 3.5.7 and includes many bug fixes (needed for mass storage mode). You should upgrade to at least 3.5.7 (or the current release of 4.0.5).