betaflight/blackbox-log-viewer #363: Release Blackbox Parser as NPM Module

Swinkid, 2019-09-04:
Hey,

Is their any plans to release just the parser as a NPM Module? If not, are others allowed to?

Thanks!

mikeller, 2019-09-04:
@Swinkid: There were no plans - until right now it appears. ;-)

I think this would be a great opportunity to make the parsing (and with it the awareness of differences in past and future versions) available for other developers and projects.
Happy to help with this, and I think it's probably best to keep this npm module inside the Betaflight project, to facilitate coordination of changes to the log file format.

Swinkid, 2019-09-05:
Any ideas on where to start and suggest project structure? Obviously want to set it up in a way that the whole project isn't published to NPM. Also might be worth trying out the new packages feature in github too.

Cheers!

mikeller, 2019-09-05:
I'd probably cut as low down as possible - create a library that can read data out of a log file, and return it as data structures.

Docteh, 2019-10-05:
@Swinkid Hacktoberfest!

Swinkid, 2019-10-05:
It's coming, i just had to finish my master's thesis. Now. Free!

Docteh, 2019-10-05:
I actually haven't given the blackbox viewer much consideration, based on file names I'd start with `js/flightlog_*`

closed 2019-11-12 by stale[bot]