betaflight/blackbox-log-viewer #369: Error opening log: No logs in this file could be parsed successfully

fremebondo, 2019-09-28:
[LOG00019.TXT](https://github.com/betaflight/blackbox-log-viewer/files/3665667/LOG00019.TXT)

Configurator: 10.5.1
Firmware: BTFL 4.0.5
Target: SPRACINGF3

# diff
# Betaflight / SPRACINGF3 (SRF3) 4.0.5 Aug  1 2019 / 12:24:55 (839661b20) MSP API: 1.41 / FEATURE CUT LEVEL 6
batch start
name FremeVTOL
resource MOTOR 8 NONE
resource SERIAL_TX 11 A03
mixer CUSTOMAIRPLANE
smix 0 0 0 100 0 0 100 0
smix 1 1 1 100 0 0 100 0
smix 2 2 2 100 0 0 100 0
smix 3 3 0 100 0 0 100 0
smix 4 4 1 100 0 0 100 0
smix 5 5 2 100 0 0 100 0

feature -RX_PPM
feature -RSSI_ADC
feature -ANTI_GRAVITY
feature -DYNAMIC_FILTER
feature RX_SERIAL
feature MOTOR_STOP
feature SOFTSERIAL
serial 1 1 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 30 128 115200 57600 0 19200
aux 0 0 6 1250 2100 0 0
aux 1 1 2 1300 2100 0 0
aux 2 13 4 1300 1700 0 0
aux 3 26 8 1300 1675 0 0
aux 4 31 8 1700 2100 0 0
set gyro_sync_denom = 2
set gyro_lowpass_hz = 50
set gyro_lowpass2_type = BIQUAD
set gyro_lowpass2_hz = 10
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 10
set acc_trim_pitch = 42
set acc_calibration = -4,-2,-36
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SBUS
set blackbox_p_ratio = 16
set blackbox_device = SERIAL
set motor_pwm_protocol = OFF
set current_meter = NONE
set vbat_scale = 10
set vbat_divider = 1
set deadband = 12
set yaw_deadband = 14
set pid_process_denom = 8
set gyro_1_sensor_align = DEFAULT
profile 0

set anti_gravity_gain = 4900
set throttle_boost = 14
set acro_trainer_angle_limit = 50
set p_pitch = 51
set i_pitch = 3
set d_pitch = 6
set f_pitch = 71
set p_roll = 40
set i_roll = 3
set d_roll = 15
set f_roll = 105
set p_yaw = 102
set i_yaw = 5
set angle_level_strength = 25
set d_min_roll = 0
set d_min_pitch = 0
set d_min_boost_gain = 23
rateprofile 0

set roll_rc_rate = 49
set pitch_rc_rate = 40
set yaw_rc_rate = 32
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 30
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 50
set tpa_rate = 0
set throttle_limit_percent = 99
batch end

#

fremebondo, 2019-09-28:
Tried moving BB on harware serial, and removing OpenLog too. Attached file is recorder with USBFTDI and RealTerm. This file is not readable too in BBExplorer
[capture.txt](https://github.com/betaflight/blackbox-log-viewer/files/3665726/capture.txt)
....

fremebondo, 2019-09-28:
I tried moving the FC and the sticks while recording... Now I can open the log but it shows almost nothing.
[capture2.txt](https://github.com/betaflight/blackbox-log-viewer/files/3665742/capture2.txt)

fremebondo, 2019-09-28:
No explanation here...
Saved in ONBOARD FLASH (no serial com errors) ... BBExplorer still shows an not-existend flight

closed 2019-11-04 by stale[bot]

Aaronprat, 2020-01-05:
@fremebondo did you find the root cause? I'm experiencing a similar issue.

fremebondo, 2020-01-07:
No, still no luck, I tried with blackbox ratio set to 1 up to 2048, I tried
lowering pid loop frequqncy, and disabling features as well. No way to get
a single readable log.

Still waiting for help...

access3424, 2020-06-11:
I have the same problem. What is going on!