betaflight/blackbox-log-viewer #372: Advances User Settings: Buttons not visible if Watermark is ON

mrRobot62, 2019-10-02:
Behaviour:
MacOS Mojave 10.14.06, MacBook Pro, 13" Retina Display
BBE: 3.4.0

Advanced User Settings:

If slider "Watermark" is ON, windows size expand on Y-Size out of screen, with result, that buttons are not longer visible. Window is not movable.

[image]

mikeller, 2019-10-02:
@mrRobot62: BTW: I built a mac dmg file for 3.4.0 - is it possible to upload this file for other mac-users.

Sure, feel free to add it here - or add a link to version shared on Google Drive or dropbox if it is too large.

We currently do not have a CI set up that is capable of building and publishing a MacOS installer for all changes to `master` - this is due to the fact that it is only buildable on MacOS, and us not having a suitable MacOS CI server.

mrRobot62, 2019-10-02:
I uploaded the dmg-file on my google-drive

[https://drive.google.com/open?id=1B5tEjiLgjrrAfoe9cAklr0UqW1_kekMt](url).

If needed, I can create dmg-files on my MacBookPro for you :-) - send me a notice -

mrRobot62, 2019-10-23:
Bug fixed in 3.5.0RC1

closed 2019-10-23 by mrRobot62