betaflight/blackbox-log-viewer #373: Feature: Give Workspaces "titles/names" and display this name as legend title

mrRobot62, 2019-10-02:
It would be helpful, if user use different workspace favorites (like UAVTech.json). To give the user a better overview its helpful if a favorites contains a title and this title is displayed on the legend canvas.

[image]

wolkesson, 2019-10-11:
I'm working on this! The goal is to also make:
- the title editable by double-click
- a dropdown for selecting workspace, same as pressing [1-9]
- A save workspace button, same a pressing shift+[1-9]
- Make it backwards compatible...

Anyone; is it better to put it on the toolbar, next to time? Possibly a workspace could be more than only a graph config. It could for example be an analyser overlay or show/hide states of stick and craft.

wolkesson, 2019-10-12:
Hacking around a bit and decided on a more elaborate solution: 
[image]

[image]

Still some work to do with backward compatibility and actually showing it in the title in the legend. Keyboard commands will be as before. 
This also solve #360 

Feedback is welcome! PR in a couple of days.

mrRobot62, 2019-10-14:
Hi @wolkesson - sorry for this late response :-( - I think your solution with a combobox is nice and give the user the possibility to change a workspace very easy. To edit names is perfect.

Good solution (for me ;-) )

mrRobot62, 2019-10-15:
Update: if you enabling "editing workspace name" please write it back into workspace json file, that would be comfortable for users

wolkesson, 2019-10-15:
Have a look at the PR #390. :D 
Workspace names are written to the json. Required a new format, but it's backwards compatible, meaning you will be able to open for example UAVTech.json. However new files will not open in older versions. A good way of testing this out is to give names to all the workspaces in UAVTech according to this description: https://drive.google.com/open?id=18a-YCHqF720qgsbw7XNrANm0juFsLNPD

mrRobot62, 2019-10-16:
ui - seems to be very comfortable :-) - will download current branch and create a dmg-file. Unfortunately not today and maybe not tomorrow :-( - If done, I will give a feedback

mrRobot62, 2019-10-23:
I testes 3.5.0RC1 with @wolkesson extension - works fine - but I found following "unsharpness" ;-)

- Workspace name length:
if to long ... dots are shown, but no chance to see (e.g. tooltip) the complete name:
Solution: max number of chars or tooltip or wider box

[image]

-

mrRobot62, 2019-11-03:
made a small enhancement for this feature.
[image]

mrRobot62, 2019-11-05:
done with #400/#401

closed 2019-11-05 by mrRobot62