betaflight/blackbox-log-viewer #383: Blackbox 3.4 Shows DSHOT Bdir as OFF When It Is Enabled

azixaka, 2019-10-09:
DSHOT is enabled which can be seen below, however header info shows is as being off.

[image]

diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.0 Oct  8 2019 / 01:30:00 (0634b1d07) MSP API: 1.42
# manufacturer_id: MTKS   board_name: MATEKF722MINI   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF722MINI
manufacturer_id MTKS
mcu_id 005200323237511534363730
signature 

# name: babomish

# resources
resource CAMERA_CONTROL 1 A01

# beacon
beacon RX_LOST
beacon RX_SET

# map
map ATER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200

# aux
aux 0 0 0 1400 1600 0 0
aux 1 1 3 1700 2100 0 0
aux 2 13 3 1400 1600 0 0
aux 3 35 1 1700 2100 0 0
aux 4 39 2 1700 2100 0 0
aux 5 40 2 900 1300 0 0
aux 6 49 1 1300 1700 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 25 200 400 600
vtxtable powerlabels 25 200 400 800

# master
set gyro_lowpass2_hz = 500
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 400
set dyn_lpf_gyro_max_hz = 1000
set acc_calibration = 104,58,281
set rssi_scale = 200
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 64
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set small_angle = 180
set pid_process_denom = 1
set osd_warn_batt_warning = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_vbat_pos = 2498
set osd_rssi_pos = 2521
set osd_throttle_pos = 2515
set osd_current_pos = 2505
set osd_mah_drawn_pos = 490
set osd_stat_rtc_date_time = ON
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_max_g_force = ON
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5917
set name = babomish

profile 0

# profile 0
set p_pitch = 65
set i_pitch = 100
set d_pitch = 33
set f_pitch = 0
set p_roll = 65
set i_roll = 95
set d_roll = 30
set f_roll = 0
set p_yaw = 60
set i_yaw = 100
set f_yaw = 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 181
set pitch_rc_rate = 181
set yaw_rc_rate = 181
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

McGiverGim, 2019-10-09:
Oh, I will look into it ☹️

Can you attach you Blackbox file?

azixaka, 2019-10-09:
Sure, it's [here ](https://1drv.ms/u/s!AiGgeJ7isHv6kpQtNWLWXnO1Ojq3Zg?e=BUZmJC)

McGiverGim, 2019-10-09:
@azixaka your log is old. Your diff is from a firmware version of yesterday but your log is from last saturday. The bidir header value was added in the firmware here: https://github.com/betaflight/betaflight/pull/8992 maybe it was merged after you flashed your quad?

azixaka, 2019-10-09:
I confirm it works with the latest log file. There is however definitely a bug in the other section:

[image]

That is definitely set to LOW in my settings that you can see above.

The new log file is at the same location in case you need it. It is the btfl_002.bbl

McGiverGim, 2019-10-09:
Ok, I will look into it tomorrow ☹️ now I'm out of the computer.

IvoFPV, 2019-10-09:
@McGiverGim It was just a range name order issue. Unfortunately 3.4.0 was released today.

McGiverGim, 2019-10-09:
Yes, that is was I was fearing... :-(

azixaka, 2019-10-09:
3.4.1 can be released tomorrow. ;)

PR #384 (2019-10-12): Fix dyn notch range
Fixes #383 
Dynamic notch range naming order was incorrect.
closed 2019-10-12 by mikeller