betaflight/blackbox-log-viewer #385: Bug: can't load video (MP4 or AVI) on MacOS Mojave: Error while loading the video. ERROR (4): DEMUXER_ERROR_NO_SUPPORTED_STREAMS: FFmpegDemuxer: no supported streams

mrRobot62, 2019-10-09:
Hi, tried to load a video in Blackbox-log-viewer 3.4.0.

Even I use mp4 file or an avi file - I got the same error message.

Setup:
Modellname:	MacBook Pro
  Modell-Identifizierung:	MacBookPro12,1
  Prozessortyp:	Intel Core i7
  Prozessorgeschwindigkeit:	3,1 GHz
  Anzahl der Prozessoren:	1
  Gesamtanzahl der Kerne:	2
  L2-Cache (pro Kern):	256 KB
  L3-Cache:	4 MB
  Hyper-Threading Technologie:	Aktiviert
  Speicher:	16 GB
  Boot-ROM-Version:	186.0.0.0.0
  SMC-Version (System):	2.28f7
  

Blackbox-Log-Viewer: 3.4.0

I found an issue based on same error message - but no solution - issue was closed.

Hope, somebody can assist me

mrRobot62, 2019-10-09:
**UPDATE** 
I found a solution for above error:
If I convert the AVI file into H.264/mpeg-4 AVEC mp4 format it's possible to load the movie file. Is it possible that the underlying video converter do not except other video formats as H.264?

IvoFPV, 2019-10-09:
@mrRobot62 Yeah, .avi from the Fatshark DVR never worked for me. I previously used to convert to .mp4 h.264 and then it worked. You can check out #344 for more info why it doesn't work.

mikeller, 2019-10-09:
Closing as duplicate of #344.

See also https://github.com/betaflight/blackbox-log-viewer/issues/344#issuecomment-496912290

closed 2019-10-09 by mikeller

mrRobot62, 2019-10-10:
Accepted - but maybe it would be helpful for other users to give a clear statement inside wiki, that movie overlays only work with mp4 H.264 format. (of if already available - I never read it :-O ). Thanks for your support ! Good job men

McGiverGim, 2019-10-10:
We use the ffmpeg library ready for node.js from here: https://github.com/iteufel/nwjs-ffmpeg-prebuilt/

So we support the formats that this library support. Not more, not less. ;)