betaflight/blackbox-log-viewer #391: Freq. vs Throttle Analyser Display Doesn't Work For Multi-Log Files

azixaka, 2019-10-16:
BF 4.1 RC6 generated log.
3.4.0 BBE

Steps to reproduce:
* Open any log that has more than one log session in it.
* Bring on the spectrum analyser display window.
* Switch to Freq. vs Throttle mode and it will get ignored.

[Log](https://1drv.ms/u/s!AiGgeJ7isHv6kpQtjcEtV3a1o0G9gA?e=fofntJ) used to reproduce this issue.

mrRobot62, 2019-10-16:
I tried to load your file into my 3.4.0 OSX version - and it works

[image]

Are you sure that u used the 2nd block and not the first block?
1st block do not contain enough data (very often if your armed/disarmed in a short period). 1st block is only 4sec - 2nd block 2:48mins

azixaka, 2019-10-16:
Yes, I'm sure I used the second block and I'm on Windows if that matters.

mrRobot62, 2019-10-16:
depending on performance of your PC and some "other" behaviors, it can take some seconds to create a heat-map (noise map).

But please check following:
1) activate spectrogram 
2) change to FREQ vs Throttle (heat map)
3) click on your graph (e.g. ROLL) again and wait several seconds.
To create a heat map it taks more time because this kind of graphs include 3 dimensions (the color is the 3rd dimension).

With your log, it took on my machine at least 15-20sec to generate this heat map 

Be patient ;-)

But to get it faster, mark relevant data with a start point (IN => I on your keyboard ) and end point (OUT => O-Key) (see screen shot. It will work faster for you. (what is your blackbox log rate? (1k,2k, ...)

[image]

azixaka, 2019-10-16:
I tried all that. I open log for another session that doesn't have two sessions and it works perfectly.

mrRobot62, 2019-10-16:
I assume this issue can be closed by admins - right?

azixaka, 2019-10-16:
I just tried it on my other laptop that didn't even have BlackBoxExplorer installed and it's the same issue. Can you try it on Windows?

azixaka, 2019-10-16:
And just tried on my mate's Windows laptop - Waited for 5 minutes. There is definitely a bug.

[image]

mrRobot62, 2019-10-16:
I built up a virtual machine on my mac with Win 7. After installing BBE 3.4.0 and loading your file I can do the same as under MAC

[image]

**BUT**
I found following things
a) first try after loading your file I saw NO graphs (uhh - not good) (like on your screenshot). Restart from BBE didn't change this.
My workaround was: click on Graph setup button - and save again - afterwards all graphs were visible.
I can't reproduce it yet, because if I know start BBE 3.4.0 again and load your file - it works inside 2-3 secs.

**Building the heat map => please strip your data with I-Key and O-Key !!!**

b) honestly your files are very big (10mb, 14MB, ...) for a bb-analyse maybe they contain not usable data. Tip (as I heard from Mark): do some short punches, quick rolls/flips, fast turns, yaw spins. Come down, start your analysis. Don't use your complete battery time for an "analyse flight" :-)

Conclusion:
On my setup (OSX and windows) both systems are working with your file(s) (I tried one of your others to. On both environments it take time to load and publish the data - I assume on huge data set.

azixaka, 2019-10-16:
Have you tried on Windows 10?

mrRobot62, 2019-10-16:
**no - I'm not able to do this** - I'm not a member of the dev-team. They have to decide this.
It was a quick analyse from my side - don't worry - my goal is to provide some tips and information to you.

Sorry I do not have a win10 VM available (I think so)

mrRobot62, 2019-10-16:
Update: I had previously installed a Win10 VM :-)
Installed BBE 3.4.0
Load your file
made below screen shot

azixaka, 2019-10-16:
Wow, no it's weird. I have bigger files (I put some in that same shared folder) and they don't have that issue at all - they just load and they are bigger. I guess it doesn't matter then.

McGiverGim, 2019-10-16:
I have reproduced it, but only sometimes... sometimes I have the error, sometimes I have the graph... I will try to search for the error when I have some more time.

McGiverGim, 2019-10-16:
I was wrong, I don't have a graph because I selected the first flight, that has no throttle.

When I select the second one, it works perfectly under Windows 10:
[image]

McGiverGim, 2019-10-16:
I don't think it takes more time if it has more than one flight. If you have samples I can look into it, I don't have this kind of files (my FCs have SD, not flash).

If you start a debug version of the Configurator (yarn gulp debug) you can debug it, all the tools are at the right click.

mrRobot62, 2019-10-17:
> BF 4.1 RC6 generated log.
> 3.4.0 BBE
>
> Steps to reproduce:
>
> * Open any log that has more than one log session in it.
> * Bring on the spectrum analyser display window.
> * Switch to Freq. vs Throttle mode [...]

These are three files from @azixaka two with one block and one with two blocks.

Last night I tried it again under my MacBoo Catalina - mhm in some circumstances I have delays if I start BBE - and than load a one-block file - it take no long time (1-2sec).
If I load load as a second file the two block file - it loads faster - weird 

If I start BBE and load at first the two-block file - sometimes it take a long time - ahh up to 20secs - weird-weird

McGiverGim, 2019-10-17:
No,I'm not available. But you can find me and a lot of other people willing to help in the official Slack chat of Betaflight.

About loading multiple block files, they take longer because it must read the ENTIRE file to get all the headers of all the blocks. So you must sum the size of all the blocks, not only the block that you have open.  

Maybe the Spectrogram is open when you select a new file? If it is it calculates all the info needed by it, so maybe this is the reason of the slow charge too.

closed 2019-11-23 by stale[bot]