betaflight/blackbox-log-viewer #406: remove gradients

meoso, 2019-11-07:
Is there a way to remove the graphical gradients? or can the source-code to produce then be removed.  it interferes with overlaying any exported video onto flights in other post-production software.  

[image]

my 3rd party software works great to overlay exported video and see the flight footage underneath, the only downfall is i see a slight variation of color due to these gradients.

wolkesson, 2019-11-15:
Faily easy to fix. Just don't know where to put the setting? 
[image]
Or, should we remove them all together? Would anyone actually miss them? I know that it is for showing signal scale, but that's what grids are for.

mikeller, 2019-11-15:
In a perfect world (i.e. without customers :laughing: ) I'd say remove. But since there are sure some to be sad about it being gone, we already have it, and the cost for making it switchable isn't going to be huge, I'd say keep it, make it switchable, and default to off.
If nobody complains about it being gone after this we can remove it in the next version. :wink: 
Position looks good to me.

mikeller, 2019-11-15:
Update: On second thought, these settings could probably do with some reorganising - they are mostly single item groups now which is a bit daft.

How about:
- Craft
-- Use custom Mixer
-- Positions
- Display
-- Show units on legend
-- Show watermark
-- Show background gradient
-- Show Lap timer
(leave Stick Settings / Analyser Settings as they are)

@McGiverGim: Thoughts?

Also, this can be done in a separate pull request.

wolkesson, 2019-11-15:
I was just going to suggest something similar. I have a pull request for this the gradient on/off feature ready. Also thinking that in Advanced Settings there are settings for show/hide lap timer and watermark, but not for show/hide of sticks and craft. These are instead controlled from the toolbar. Shouldn't we move lap timer and watermark to the toolbar too? Maybe making the overlay into a dropdown (not only for smaller screens) and putting all that has to do with visibility here?

I suggest a new issue for restructuring settings dialog.

mikeller, 2019-11-16:
@wolkesson: Yes re adding these show / hide settings to the menu - their discoverability at the moment is horrid.

McGiverGim, 2019-11-16:
To me is ok. I think a reorder is good.

meoso, 2019-11-16:
i could be wrong, but this never did anything that i could see.. maybe toggle it with this icon:
[image]

in the meantime, i just changed  `rgba(255,255,255,0.1)');` to `axisGradient.addColorStop(0.0,   'rgba(0,0,0,0)');` in `js/grapher.js` lines 413,417 and compiled.

wolkesson, 2019-11-16:
For the grid button usage see #407 
For the fix of gradient there is a pull request #408 
🕹

PR #408 (2019-11-17): Setting to show/hide background gradient off graph.
Closes: #406
closed 2019-11-17 by mikeller