betaflight/blackbox-log-viewer #409: Move Analyser into a separate view

wolkesson, 2019-11-16:
I don't see the use of having the Analyser (frequency plot) overlaying the graphs. This does not relate to the current time in the graph plot, only to the in and out markers. I makes more sense in my opinion to have a separate view button for this (like the table view) and always having it show in "full screen". More preset "Type of Spectrum" could be created like the "Freq vs Throttle"

McGiverGim, 2019-11-16:
I'm not too sure, if the graph refresh with the in and out maybe it is easier to show it in the same window to do it easily. Some Blackbox expert users required here, like @ctzsnooze to give his opinion or @spatzengr

wolkesson, 2019-11-29:
The log-seek-bar will still be visible, so setting in and out will still be possible. It will be like the Analyzer in full screen, but without the semi-transparent background. What will be lost is the possibility to have the Analyzer as a picture-in-picture overlay when rendering to video.

McGiverGim, 2019-11-29:
It seems @mikeller is ok with this (thumb up emoji), so if you are ok with the change, go for it. 
I like to have the analyzer with the graph because sometimes I see some noise in the graph, with the mark (M) I look what frequency is, and I try to look for it in the spectrum to see if it is a noticeable point or not.
Nobody has commented so I let this as you convenience... maybe we have complains after merged... 😋

closed 2020-01-05 by stale[bot]