betaflight/blackbox-log-viewer #412: 3.5.0RC2 - OSX dmg file available

mrRobot62, 2019-11-17:
a new OS X file is available.

Tested on OSX-Catalina works fine

Download file here: 
https://drive.google.com/file/d/15zzq5jl7QAjGt6u59au9ePahaJXqHYtr/view?usp=sharing

mikeller, 2019-11-18:
@mrRobot62: Can you please not call these `-RC<x>`? If we decide we are going to need release candidates for 3.5.0 this will be confusing. You could name them `-RC-mrRoboto62-2` or `-mrRoboto62-1` or similar.

Also, we now have CI generated nightly builds for all platforms available at https://github.com/betaflight/blackbox-log-viewer-nightlies/releases, so this is probably not needed any more.

mrRobot62, 2019-11-18:
Hi @mikeller  - oh that was unknown for me - sorry - if will close this issue.

closed 2019-11-18 by mrRobot62