betaflight/blackbox-log-viewer #418: Export gyro data to GoPro GPMF file for reelsteady

jacobbloy, 2019-11-26:
Export gyro data to GoPro GPMF file for reelsteady, this would allow us to use our turtle or runcam split camera or DJI FPV system HD recording and input them into the reelsteady go software and get supper smooth camera footage with out the need for a GoPro. Reelsteady said the emu flight guys are looking into making a tool.

mikeller, 2019-11-27:
I think this can be done already with existing tools: Use https://github.com/betaflight/blackbox-tools to convert the Betaflight log into CSV, then use a tool to create the GPMF file.

meekaah, 2019-12-18:
i've thought about this alors in the past few months, this would be so amazing to be able to get reelsteady results with the dji fpv or any other non gopro cameras

meekaah, 2020-01-19:
nobody willing to give this a shot? i'll try and do a proof of concept see if it can work..

closed 2020-02-25 by stale[bot]

jbstand, 2020-03-25:
@meekaah did you do the proof of concept? Can I help?

meekaah, 2020-03-27:
> @meekaah did you do the proof of concept? Can I help?

no I havent had the time off work to get this done, it would be amazing though

sansha, 2020-05-09:
GoPro actually open-sourced the code to write these files: https://github.com/gopro/gpmf-write
So it would probably be possible to convert the blackbox files to gpmf. The questions is, however, how one could sync the video & telemetry.

mikeller, 2020-05-10:
@sansha: Synchronisation with the blackbox log could probably be done through the arming beeps recorded on the video's audio track.

sansha, 2020-05-13:
That's true. I don't have a beeper but the motor noise itself is probably sufficient. Unfortunately I don't have time to look into it right now but I guess it would be a standalone script and not implemented into betaflight, right?

mikeller, 2020-05-14:
@sansha: It should probably be a standalone project, potentially using the data access library that we have in the blackbox tools.

dngrlux, 2020-06-04:
I've dreamt of this since early 2015, long before ReelSteady GO and GoPro telemetry. Would be a great addition! https://github.com/cleanflight/blackbox-log-viewer/issues/9

kingfisher09, 2020-07-27:
Hi, have you guys made any progress on this? I'd be happy to help if I can although my abilities are largely limited to python.

ElvinC, 2020-08-10:
I've been working on a project for stabilizing footage using gyro data. The goal is to make something that works with data from the Betaflight Blackbox log and non-GoPro cameras. While stabilization itself is still work in progress, you can check out the current code [here](https://github.com/ElvinC/gyroflow). My plan is to focus on GoPro footage with embedded gyro data first, and then move over to Blackbox data. If anyone is interested in the details, I wrote a post [here](https://redd.it/i6ohqc) about the current status of the project.

mooiweertje, 2021-01-19:
It would be nice if the log would only contain gyro data so the log file uses less space. My internal flash is only 16MB.

McGiverGim, 2021-01-20:
> It would be nice if the log would only contain gyro data so the log file uses less space. My internal flash is only 16MB.

I think that what you want was done here: https://github.com/betaflight/betaflight/pull/9726

mooiweertje, 2021-01-20:
Ah yes, but not part of 4.2.6 but 4.3