betaflight/blackbox-log-viewer #423: Add legend to the analyser display

mikeller, 2019-12-26:
Currently, when the analyser display is opened, there is no indication of what data stream was used as input for the analyser. This can be confusing, especially when dealing with screenshots:

[image]

vs.

[image]

A legend should be added to the analyser display to indicate the data stream that the analysis was based on.

McGiverGim, 2019-12-26:
This has been broken recently. The name must appear at the bottom right of the analyser. This only happens at full screen?

mikeller, 2019-12-26:
@McGiverGim: You are right, the title is there:

[image]

It is just very hard to spot behind the green plot - maybe it should go somewhare else like up top centered where it is less likely to be overlaid?

@hydra: Fyi.

McGiverGim, 2019-12-27:
Usually the noise is almost zero there, so it is more visible. Maybe adding a black border to the letters will make them clearer. I will play a little with it when I have some time.

closed 2020-06-29 by mikeller