betaflight/blackbox-log-viewer #426: Having workspace issues on Apple / Mac

exocode, 2019-12-30:
I use this workspace file https://drive.google.com/drive/folders/1hWgcADCI3Aa4XLUiGsGQbPHxdkE94taP taken from UAV-Tech).
* loaded via "open log file/video)
* get "workspace loaded" modal
* pressing 1-9
* nothing happens
* tried to export workspaces
* exported workspace.json matches exactly to that what I imported

I am using a Mac (Mojave 10.14.6 (18G103)). My logs are recorded from Rev : Betaflight 4.1.2 (89051e256) STM32F405
Blackbox Explorer is v3.4.0

any suggestions? is this my fault?

boscomonkey, 2020-01-05:
I have the exact same setup on a 2015 Macbook Pro 13" Retina - Mojave 10.14.6, BF 4.1.2, BBE 3.4.0 - with the same issue: shortcut keys 1-9 do not work.

exocode, 2020-01-05:
After some trial and error I could get it working by using an old template:
It seems to be a bug!

1. remove all old preferences (not sure if nessessary)
`rm ~/Library/Preferences/com.nw-builder.betaflight-blackbox-explorer.plist`
2. Open most recent version (3.4.0) 
3. Open a valid BBL file
4. Remove all graphs and hit save
5. Open this file  [UAVtech BetaFlight Blackbox explorer template settings.json.zip](https://github.com/betaflight/blackbox-log-viewer/files/4023280/UAVtech.BetaFlight.Blackbox.explorer.template.settings.json.zip)
6. Nothing seems happened
7. Hit the number keys on your keyboard (the graphs strangely appear now)
This wont work with the file in my link above
--
8. Remove all graphs again, hit save
9. Hit the numbers on your keyboard (the graphs strangely appear again)
10 repeat step 8+9 😂

boscomonkey, 2020-01-05:
Nice sleuthing! I was able to repeat your results.

Removing the preferences was not necessary. The [UAVtech BetaFlight Blackbox explorer template settings.json.zip](https://github.com/betaflight/blackbox-log-viewer/files/4023280/UAVtech.BetaFlight.Blackbox.explorer.template.settings.json.zip) to which you linked in Step #5 works fine (after unzipping).

There must be a bug in UAVTech's settings JSON file that's in his Google Drive.

boscomonkey, 2020-01-06:
UAVTech's settings JSON doesn't work in Windows 10 either. I did a side-by-side comparison of the older settings file you found with what's on his Google Drive, and discovered that his JSON file is slightly "inverted".

Instead of being a hash key "graphConfig" mapped to an array of shortcut definitions, his file is an array of hashes, each of which has a "graphConfig" mapped to shortcut definitions. That's pretty easy to fix, and I wrote a quick & dirty Ruby script to turn his file into something importable into BBE.
* https://github.com/boscomonkey/fix_uavtech_json

Attached is the fixed settings file for your convenience: [fixed-settings.json.zip](https://github.com/betaflight/blackbox-log-viewer/files/4024656/fixed-settings.json.zip)

Note that you'll need to unzip it before importing into BBE. To be safe, you should really look over the Ruby script to make sure it's legit, and run it yourself instead of trusting some random uploaded zip file.

wolkesson, 2020-01-06:
I could be my fault! When I did the new workspace feature, the format was changed. It was supposed to be able to open the old format, but I sure could have missed something... Unfortunately, I don't have time to look closer into it at the moment. Maybe next weekend. Sorry!

NeuerUser, 2020-01-13:
Yes, the problem is that the format has apparently changed. BBE 3.4 is not able to read the old format.

exocode, 2020-01-13:
> Yes, the problem is that the format has apparently changed. BBE 3.4 is not able to read the old format.

Are you using mac? Try to load this one: 
(it's just a fresh exported simple,default settings snippet)
```
[
    {
        "graphConfig": [
            {
                "fields": [
                    {
                        "color": "#fb8072",
                        "curve": {
                            "inputRange": 916,
                            "offset": -1057,
                            "outputRange": 1,
                            "power": 0.25
                        },
                        "default": {
                            "outputRange": 1,
                            "power": 0.25,
                            "smoothing": 0
                        },
                        "grid": false,
                        "lineWidth": 1,
                        "name": "debug[0]",
                        "smoothing": 0
                    },
                    {
                        "color": "#bebada",
                        "curve": {
                            "inputRange": 1199.9999999999998,
                            "offset": 0,
                            "outputRange": 1,
                            "power": 0.25
                        },
                        "default": {
                            "outputRange": 1,
                            "power": 0.25,
                            "smoothing": 3000
                        },
                        "grid": false,
                        "lineWidth": 1,
                        "name": "gyroADC[0]",
                        "smoothing": 3000
                    },
                    {
                        "color": "#b3de69",
                        "curve": {
                            "inputRange": 1199.9999999999998,
                            "offset": 0,
                            "outputRange": 1,
                            "power": 0.25
                        },
                        "default": {
                            "outputRange": 1,
                            "power": 0.25,
                            "smoothing": 0
                        },
                        "grid": true,
                        "lineWidth": 2,
                        "name": "rcCommands[0]",
                        "smoothing": 0
                    },
                    {
                        "color": "#fdb462",
                        "curve": {
                            "inputRange": 50,
                            "offset": -50,
                            "outputRange": 1,
[... 2589 lines cut ...]
                        "curve": {
                            "inputRange": 1000,
                            "offset": 0,
                            "outputRange": 1,
                            "power": 0.3
                        },
                        "default": {
                            "outputRange": 1,
                            "power": 0.3,
                            "smoothing": 3000
                        },
                        "grid": true,
                        "lineWidth": 1,
                        "name": "axisSum[0]",
                        "smoothing": 3000
                    }
                ],
                "height": 1,
                "label": "PIDs"
            }
        ],
        "title": "Roll Details"
    }
]
```

NeuerUser, 2020-01-13:
> Are you using mac? Try to load this one:
> (it's just a fresh exported simple,default settings snippet)
> 
I'm on Linux. The above posted fixed file works for me.
Thanks!

exocode, 2020-01-13:
It'n NOT WORKING FOR ME :-(

NeuerUser, 2020-01-13:
This one worked for me:
https://github.com/betaflight/blackbox-log-viewer/files/4023280/UAVtech.BetaFlight.Blackbox.explorer.template.settings.json.zip

exocode, 2020-01-13:
> This one worked for me:
> https://github.com/betaflight/blackbox-log-viewer/files/4023280/UAVtech.BetaFlight.Blackbox.explorer.template.settings.json.zip

Yes, that is one that I created and posted. **But it is not working on a MAC** (this is the reason for this open issue :-)

boscomonkey, 2020-01-21:
For your convenience, I made a new commit to https://github.com/boscomonkey/fix_uavtech_json that adds:
* UAV Tech's original file that doesn't import properly (dated 2019/12/27).
* The fixed version of the above.

Have fun.

closed 2020-02-27 by stale[bot]