betaflight/blackbox-log-viewer #427: bug: pressing H quickly & repeatedly will fade the background completely to black

nerdCopter, 2020-01-12:
Pressing H, waiting for the header display, then pressing H again work fine; however, pressing H quickly multiple times will gradually (or not-so gradually) darken the main interface completely without recovering to normal opacity.

`v20191223.80`

`Linux 5.0.0-37-generic #40-Ubuntu SMP Thu Nov 14 00:14:01 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

McGiverGim, 2020-03-25:
It seems that this is a bug produced by the `bootstrap.js` library. Is used to show and hide the modal dialogs.

When we press the `H` very fast, the event that "shows/hides" the dialog does not remove the fade div that is in the background, but adds another one in top of it:

[image]

I have tried to update the library without luck, the bug persist, and I'm searching for a workaround for this... someone has experience with this library?

mikeller, 2020-03-25:
Good find, @McGiverGim!

McGiverGim, 2020-03-25:
I have found a workaround... I will push a PR in some minutes... let's see if it does not produce other glitches...

closed 2020-03-26 by mikeller