betaflight/blackbox-log-viewer #438: Actual rates gyro and setpoint scales incorrect

Justicefpv, 2020-04-12:
When using Actual Rates blackbox viewer gyro lines look stepped and full stick deflection looks like a dashed line

ctzsnooze commented in slack that to fix this Gyro and setpoint scales need to be set to 20%

[image]

McGiverGim, 2020-04-12:
This is because it calculates the scale using the max rate of the old rate system. It needs to take into account the new system. Is this info (the system that is being used) in the blackbox header? Can you attach your log file as sample?

ctzsnooze, 2020-04-12:
I had a quick look at the code, it calculates setpoint from the Betaflight rates maths, and isn't aware of any other Rates types.
Setpoint will be incorrect for all the others.
Unfortunately support for other rates types hasn't been implemented in BBE previously so it would require changes to logging that included the rates type, and then to the maths that calculated the rates.
Quite a lot of work, unfortunately.
Also the new deadband maths is a little different now.
There is js maths in Configurator that exactly calculates all current 4.2 rates methods, including deadband adjustments.
In the meantime, 'Actual' rates around centre stick will be reasonably comparable to Betaflight only if the user manually reduces the scale factor in BBE to 20%.  Simple' rates uses the same RC_Rate value as Betaflight so is the same around centre stick as Betaflight.  However, both of them will differ from betaflight rates at higher stick angles because the expo curves are different.

ctzsnooze, 2020-04-12:
Here is a dropbox link to a log with Actual rates being used.

0crap, 2020-04-13:
Yep, noticed this as well with "Quick Rates".

fgiudice98, 2020-04-14:
Actually from BF 4.0 the setpoint values are in the log itself, so they are always correct. The only problem here is the gyro scaling. It's not a big deal to make it works with newer rates, but i think is better to fix the gyro scaling at 2000dps, as it's always the max rate supported by BF itself (mostly for the gyro capabilities) and most of the users have their max rates around 1000dps

0crap, 2020-05-27:
I don't think it's a good idea to let the bot close this one.
This issue needs to be fixed.
@McGiverGim @mikeller can you please reopen and make sure the stale bot can't close it again?
Thx!

McGiverGim, 2020-05-27:
@0crap in theory, it was fixed here: https://github.com/betaflight/blackbox-log-viewer/pull/440

Did you test the latest 3.5.0 version of the Blackbox Viewer?

0crap, 2020-05-27:
Hmmm, I run into this yesterday on the 3.5.0 stable release.
I'll try to find the logfile that gives me issues.

0crap, 2020-05-27:
OK, seems I'm wrong on this, sorry for the confusion.

It seems I have a corrupt BBL file.
The header shows a red box with Rates type Betaflight, while I use QuickRates.

[image]
[image]
[LOG00111.BFL.txt](https://github.com/betaflight/blackbox-log-viewer/files/4687653/LOG00111.BFL.txt)

If you remove the .txt from the attached BBL I think you can confirm this.
If so, please close this issue again.
Thx!

McGiverGim, 2020-05-27:
Your log is from 13 May, the Betaflight firmware need this PR to add the info to the headers:
https://github.com/betaflight/blackbox-log-viewer/pull/439 and it was merged the 19 Apr.

So I close it. If you find a problem with some recent firmware you can reopen it again.

0crap, 2020-05-27:
@McGiverGim 13 May is the firmware date, which is the date BF 4.2.0 RC2 is released.
The logfile was created yesterday. (26 May)

So all recent stuff!

McGiverGim, 2020-05-27:
True, I don't know what I was thinking...

The log does not contain the `rates_type` header. If you open it with Notepad++ (or any editor) you will see it is missing. I open this again until we find the culprit...

McGiverGim, 2020-05-27:
@0crap it is a corrupted blackbox file:

[image]

The headers did not end, in theory @etracer65 fixed this in the firmware some time ago...

McGiverGim, 2020-05-27:
This is the PR: https://github.com/betaflight/betaflight/pull/8284

Maybe we have more than 3K of dat now? @mikeller @etracer65 what do you think?

But this is a new issue for the firmware, not a problem in the Blackbox Viewer. @0crap Can you open an issue there with the info? Thanks!

0crap, 2020-05-27:
Yep, that seems it.
From a different logfile I see that a small part is missing.

H airmode_activate_throttle:25
H serialrx_provider:9
H use_unsynced_pwm:0
H motor_pwm_protocol:6
H motor_pwm_rate:480
H dshot_idle_value:550
H debug_mode:6
H features:268764296
H rc_smoothing_type:1
H rc_smoothing_debug_axis:0
H rc_smoothing_cutoffs:0, 0
H rc_smoothing_auto_factor:10
H rc_smoothing_filter_type:1, 1
H rc_smoothing_active_cutoffs:54, 90
H rc_smoothing_rx_average:8356
H rates_type:4

Guess the header info keeps growing?
Maybe it's just a fluke, I don't know.

I'll keep an eye on it, if it happens more often I'll create a BF issue.

McGiverGim, 2020-05-27:
The size of this header file is more than 3.5Kb, so it seems we are going to start to get this kind of reports...

0crap, 2020-05-28:
@McGiverGim Can you have a look at the attached BBL logfile please?
The header seems fine and is not truncated, the BB logviewer is detecting QuickRates which is correct.
However I still got dotted lines on full stick deflection. Weid...
[image]
[image]
[LOG00112.BFL.txt](https://github.com/betaflight/blackbox-log-viewer/files/4695985/LOG00112.BFL.txt)

0crap, 2020-05-28:
https://github.com/betaflight/betaflight/issues/9856

0crap, 2020-05-28:
> This is the PR: [betaflight/betaflight#8284](https://github.com/betaflight/betaflight/pull/8284)
> 
> Maybe we have more than 3K of data now? @mikeller @etracer65 what do you think?
> 
> But this is a new issue for the firmware, not a problem in the Blackbox Viewer. @0crap Can you open an issue there with the info? Thanks!

Done.
https://github.com/betaflight/betaflight/issues/9856

McGiverGim, 2020-05-29:
@0crap I'm looking at it. Can you confirm what is your max rate configured in Betaflight?

In the Configurator/Blackbox if I'm not wrong, you have configured quick rates and a max degrees per second of 600/610/450 (roll, pitch, yaw).
Is this correct?
If it is, then the firmware part is different because you are making rolls at 800 deg/sec.
So one of both is not right.
Can you confirm your data in the Configurator and/or post a `diff` of your rates part?

McGiverGim, 2020-05-29:
Looking at the firmware code, it seems that the max rate is the MAX of RC RATE and SUPER RATE, but in the Configurator and Viewer is taken only from SUPER RATE.
https://github.com/IllusionFpv/betaflight/blob/fbcdeb5be6c09547c76f9177d9c0f5f856d5eacc/src/main/fc/rc.c#L210-L211
At least the Blackbox part was done for @IllusionFpv that is the developer who did the firmware part.
@IllusionFpv can you confirm if the max rate for Quickrates is taken from the max of both? If true, then the Configurator/Blackbox need a little patch. And is strange that the max dps can be greater than the max configured.

0crap, 2020-05-29:
```
# rateprofile 0
set rates_type = QUICK
set roll_rc_rate = 82
set pitch_rc_rate = 82
set yaw_rc_rate = 50
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 60
set pitch_srate = 61
set yaw_srate = 45

rateprofile 1

# rateprofile 1
set rates_type = QUICK
set roll_rc_rate = 82
set pitch_rc_rate = 82
set yaw_rc_rate = 50
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set pitch_srate = 71
set yaw_srate = 45

rateprofile 2

# rateprofile 2
set rates_type = QUICK
set roll_rc_rate = 82
set pitch_rc_rate = 82
set yaw_rc_rate = 50
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 80
set pitch_srate = 81
set yaw_srate = 45
```

I usually fly the whole flight in # rateprofile 2 and land using # rateprofile 0.
Sometimes my Taranis switch is in the position of # rateprofile 0 during take off, I then quickly switch to # rateprofile 2. (I have my Modes switch combined with a Rate Profile Selection.)
If that happened in this flight I can't remember.

So in other words 99% of the flight is done with 800/810/450 for R/P/Y.

McGiverGim, 2020-05-29:
Problem found, you have changed the rate system in flight but all the values are calculated from the data when starting the Blackbox (usually when arming)

0crap, 2020-05-29:
Nice work troubleshooting!

McGiverGim, 2020-05-29:
There is nothing we can do about this. The header gives the data to the Blackbox and we use it to prepare all the Blackbox "configuration". If something changes in flight we don't know until we arrive to this part, so there is nothing to fix here.

0crap, 2020-05-29:
Hmmm, that is a bit sad.
What do you mean exactly with "...until we arrive to this part,"?

McGiverGim, 2020-05-29:
We do not process "all the log" when loading it. We only process the part needed. When the user goes forward in the timeline we read and process the next part. So we don't know nothing about rate changes until the user goes to this part in the graph. And the scale of the graph is loaded (and not changed) when the log file is loaded.
Plus, the profile rate change event writes the new profile number used, but I doubt it writes the new values, so we don't know what are this values because they are not at the header.
The solution: in this cases you can adjust the scale manually in the graph. Fix this in software is a change complicated and at this moment I don't have time to look at it. Maybe other at some moment can do it.

fgiudice98, 2020-05-29:
As i said earlier in this issue, a solution can be to fix the scale to 2000dps. There is always the expo and the raw values, so i don't think it's a problem the fixed scale

McGiverGim, 2020-05-29:
> As i said earlier in this issue, a solution can be to fix the scale to 2000dps. There is always the expo and the raw values, so i don't think it's a problem the fixed scale

As an option maybe, but I don't like it by default at least for my use. I prefer maximum resolution and I use about 700-800 dps, so it is far from the 2000 dps suggested.

0crap, 2020-05-29:
Why not scan the rate profiles during startup and pick the highest numbers to set the scale for the BBL?
Exclude the default rate profiles so you only take the user settings into account.

McGiverGim, 2020-05-29:
Again, as an option maybe. But this will need a change in the firmware side too because only the selected rate profile information is write in the log file.

etracer65, 2020-05-29:
> Again, as an option maybe. But this will need a change in the firmware side too because only the selected rate profile information is write in the log file.

Not really possible without a huge amount of work.

closed 2020-07-25 by stale[bot]