betaflight/blackbox-log-viewer #446: I see no graphs

nablabla, 2020-06-05:
am i doing it wrong?
every info you need shoud be in the log and the screenshot

nablabla, 2020-06-05:
[btfl_001.zip](https://github.com/betaflight/blackbox-log-viewer/files/4735230/btfl_001.zip)

nablabla, 2020-06-05:
[image]

nablabla, 2020-06-05:
> 
> 
> Issue Label Bot is not confident enough to auto-label this issue. See [dashboard](https://mlbot.net/data/betaflight/blackbox-log-viewer) for more details.

what are you talking about, i was not done creating the issue, you have been to fast, sorry

spatzengr, 2020-06-05:
Looks fine to me.

[image]

Is this a build version of BF BBE or an official release you are using?

nablabla, 2020-06-05:
code > releases > Blackbox Log Viewer 3.5.0 on windows 10 (i am an expert uc;)

nablabla, 2020-06-05:
all the guys in tutorial videos just open the file with this application and they hav graphs as you have

Asizon, 2020-06-05:
O yes, i think it is happens the first time that you open bb explorer after install, just remove current graphs and reads it

nablabla, 2020-06-05:
i open and closed it several time, problem stays.
"just remove current graphs and re-add it" what is that?

Asizon, 2020-06-05:
[image]
Enter in graph setup, remove existing graphs types, and re add

nablabla, 2020-06-05:
okay, that fixed it. so it is a bug i guess,
thanks for ya help :)

spatzengr, 2020-06-05:
@nablabla , you can get trace setup Templates from tiny.cc/uavtech --> BBE Trace Templates folder too if you like.  https://www.youtube.com/watch?v=fCs51RQmxqU

nablabla, 2020-06-05:
> @nablabla , you can get trace setup Templates from tiny.cc/uavtech --> BBE Trace Templates folder too if you like. https://www.youtube.com/watch?v=fCs51RQmxqU
i have watched that already, and i love it, thanks :)

closed 2020-07-12 by stale[bot]