betaflight/blackbox-log-viewer #447: More useful default settings

0crap, 2020-06-10:
By default the BB log viewer uses Expo ON, Smoothing ON. (Also Grid ON, but I never seen a difference on that setting.)
This gives a unusual/unrealistic view of what happens and is the first thing I always switch off after loading a logfile.
[image]
When setting Expo and Smoothing to OFF it looks much more realistic.
[image]

In a rare case traces are on top of each other and I want to zoom in on what happens, I switch ON Expo.
For better usability and less mouse clicks it would be nice if the defaults can be changed to OFF.
(Maybe remove Grid, never seen that do anyting.)

Your suggestions to discuss this are welcome! :-)

ctzsnooze, 2020-06-10:
Agree!  It would be great if by default they were off.

0crap, 2020-06-12:
@McGiverGim Can you please make the PR? Then we'll see how it goes from there.

McGiverGim, 2020-06-12:
This weekend I'm out of the computer. If I get some time I will do it next week.

mikeller, 2020-06-14:
@McGiverGim: No objections from my side. I think the objections were around the fact that the current display mode for this was not visible from screenshots in earlier versions, resulting in misleading screenshots being posted.

closed 2020-06-21 by mikeller

0crap, 2020-06-21:
@McGiverGim I just installed the latest Configurator nightly build on top of my existing Windows installation. Guess what.... I see now Expo/Smoothing/Grid is automagically switched to Off.

Awesome!