betaflight/blackbox-log-viewer #471: Loading as unpacked extension in Chrome no longer works properly

James-T1, 2020-09-11:
Loading Blackbox log viewer as an unpacked extension in Chrome per the Readme no longer works properly.  I did a git bisect to find the commit where it broke, and the commit where it quit working was this one (ec4a79995f3ee91ad8f0ef5c2612d5be942c4c74):  https://github.com/betaflight/blackbox-log-viewer/commit/ec4a79995f3ee91ad8f0ef5c2612d5be942c4c74

Screenshot of working commit previous to that:  
[image]

Screenshot of non-working commit:
[image]

Using Google Chrome Version 85.0.4183.83 (Official Build) (64-bit)

Any ideas on how to get it functional as an unpacked extension again?

McGiverGim, 2020-09-22:
The theory is that the Chrome extension will get removed soon (Google is removing the support for Chrome apps).

Are you executing a `yarn install` to download the dependencies before loading it as extension?

mikeller, 2020-09-22:
@McGiverGim: Yeah, my thoughts too - Google will be retiring support for Chrome Web apps soon, so in this case I think the 'fix' is to move forward and remove support for building the Chrome Web app altogether (applies to configurator as well).

mikeller, 2020-11-07:
Superseded by #474.

closed 2020-11-07 by mikeller